Klipper3d/klipper #3185: Delta manual bed levelling - how do I get it to probe more points?

teikjoon, 2020-08-14:
Hello,

I've added the following [bed_mesh] config section

```
[bed_mesh]
speed = 50
horizontal_move_z = 5
mesh_radius = 110
round_probe_count = 9
algorithm = bicubic
```

But when I do **DELTA_CALIBRATE METHOD=manual** it only seems to probe the outer edges of the bed. Is it possible to get the manual calibration method to probe the middle of the bed?

Thanks in advance.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5072407/klippy.log)

andri-ys, 2020-08-14:
AFAIK the `bed_mesh` leveling is different from `delta_calibrate`, so you can't change any behavior of `delta_calibrate` using config for `bed_mesh`.

Isn't the `delta_calibrate` already probe the center of the bed by default?

teikjoon, 2020-08-14:
> AFAIK the `bed_mesh` leveling is different from `delta_calibrate`, so you can't change any behavior of `delta_calibrate` using config for `bed_mesh`.
> 

Doh...for some reason I thought it would generate a bed mesh to use because it put these lines in my printer.cfg :+1: 

#*# [delta_calibrate]
#*# height0 = 0.0
#*# height0_pos = 23332.000,23295.000,23442.000
#*# height1 = 0.0
#*# height1_pos = 28737.000,28732.000,20708.000
#*# height2 = 0.0
#*# height2_pos = 22839.000,32013.000,22943.000
#*# height3 = 0.0
#*# height3_pos = 20703.000,27737.000,27936.000
#*# height4 = 0.0
#*# height4_pos = 22625.000,22576.000,29217.000
#*# height5 = 0.0
#*# height5_pos = 26973.000,20802.000,27113.000
#*# height6 = 0.0
#*# height6_pos = 30411.000,22692.000,22819.000
#*# distance0 = 64.16
#*# distance0_pos1 = 23145.607,23362.179,23516.171
#*# distance0_pos2 = 21141.053,25813.865,25995.311
#*# distance1 = 63.83
#*# distance1_pos1 = 23247.853,23157.507,23619.369

> Btw isn't the `delta_calibrate` already probe the center of the bed by default?
> 

Yeah it does, but it only probes center and outer edges of the print bed :(

> To have `delta_calibrate` probe more points, as far as I know we could do it by updating the python code directly (at least that's what I did :-)

Do you have the diff for this?

andri-ys, 2020-08-14:
> Doh...for some reason I thought it would generate a bed mesh to use because it put these lines in my printer.cfg :+1:
>
> #*# [delta_calibrate]
> #*# height0 = 0.0
> #*# height0_pos = [...]

It does seems similar (they both probe the bed), yet they are different. 

I personally don't use any leveling because my calibration is already good enough (stdev around 0.03, sufficient for my finest first layer height at 0.16).

> Do you have the diff for this?

You can check [my `master` branch] (https://github.com/andri-ys/klipper). Btw it also contains code to display the stdev of the calibration. If it is already good enough, imho you could safely skip any further leveling.

teikjoon, 2020-08-15:
Thanks @andri-ys, using your helpful tips I have modified delta_calibrate.py :+1: 

```
        # Calculate default probing points
        radius = config.getfloat('radius', above=0.)
        points = [(0., 0.)]
        scatter = [.95, .90, .85, .70, .75, .80]
        for j in [0.33, 0.66, 1]:
            for i in range(8):
                r = math.radians(360 / 8 * i)
                dist = radius * j * 0.95
                points.append((math.cos(r) * dist, math.sin(r) * dist))
        self.probe_helper = probe.ProbePointsHelper(
            config, self.probe_finalize, default_points=points)

```

Which basically makes it probe 8 points each in 3 circles of increasing size (like pizzas!)

I have skipped the scatter...but I guess if I really wanted to I could import the random library...

Hywelmartin, 2020-08-16:
why???

mesh is to compensate for a bad bed surface BED_MESH_CALIBRATE
delta calibrate is, like it says, calibration for the delta kinematics DELTA_CALIBRATE

andri-ys, 2020-08-17:
Hi @Hywelmartin 

I do aware about the difference between bed mesh and delta calibrate.

As for the why, depends on the use case, I think it does useful if we could configure the probe points for `delta_calibrate` process. Adding more points could increase the quality, while using less points could speed up the process.

KevinOConnor, 2020-08-18:
> I have skipped the scatter...

FYI, the scatter is important to get good diversity on height measurements for all three towers.

That said, it's not clear why you want additional points.  The default setup provides 7 unique distances for each tower, which is enough to calibrate them.

-Kevin

teikjoon, 2020-08-18:
> > I have skipped the scatter...
> 
> FYI, the scatter is important to get good diversity on height measurements for all three towers.
> 
> That said, it's not clear why you want additional points. The default setup provides 7 unique distances for each tower, which is enough to calibrate them.
> 
> -Kevin

Yes, I seem to have confused bed mesh levelling with Delta Calibration...I've reinstated the existing code, but could I ask that the Delta Calibrate documentation make some mention of manual bed mesh levelling too? Thanks.

https://github.com/KevinOConnor/klipper/blob/master/docs/Delta_Calibrate.md

Hywelmartin, 2020-08-19:
"mesh" isn't mentioned at all on that page.. manual probing yes

closed 2020-09-17 by github-actions[bot]