Klipper3d/klipper #3187: Octoprint Menu LCD Screen

schnello, 2020-08-14:
Hi.

Since the latest update the Octoprint menu is "broken"

If the printer is in idle:
[image]

If the printer is active:
[image]

Any idea @mcmatrix?

schnello, 2020-08-14:
Ok i found the problem but i dont understand why.

For better maintance i create a local copy of the user_menu.cfg and add this in my printer.cfg
[include klipper-config/user_menu_custom.cfg]

This works well in the past. But since the latest commit the octoprint menu item is broken. All other items are ok. If i fall back to the original user_menu.cfg all is fine. (user_menu_custom.cfg is atm a 1:1 copy, no modifications done yet)

mcmatrix, 2020-08-14:
Hi @schnello 
Are these duplicate entries real menu items or remnants on the screen (can you select them)?

Could you share your full printer config.

schnello, 2020-08-14:
Attached the full config.

>Are these duplicate entries real menu items or remnants on the screen (can you select them)?

Atm iam not at home but i will try it after work.
[klipper_config_schnello.zip](https://github.com/KevinOConnor/klipper/files/5073253/klipper_config_schnello.zip)

mcmatrix, 2020-08-14:
Please try PR #3189
It should solve your issue.

schnello, 2020-08-14:
Thanks for the quick fix. Looks good.
[image]

closed 2020-08-14 by schnello

schnello, 2020-08-15:
@mcmatrix 
Onyl as gen question:
Is there a reason why in the main user_menu.cfg files the printer var are not used in the movement section:
input_max: 200
vs.
input_max: {printer.configfile.config.stepper_y.position_max}