Klipper3d/klipper #3188: uc1701 display blank screen

saurabhceaseless, 2020-08-14:
Hello

Im trying to configure a uc1701 display with my printer.

I have two mcu's attached to this system

The first one is a fysetc cheetah board running all the printer mechanics

The second one is an arduino uno which I plan to use for peripherals like display, controlled fans, etc.

Im attaching the uc1701 display to this arduino uno for which you can find the configuration below, no errors are observed in the klippy log

When the system starts (or restarts), the display does resets but ends up blank.

[mcu]
baud: 250000
serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
restart_method: cheetah

[mcu uno]
serial: /dev/serial/by-id/usb-Arduino__www.arduino.cc__0043_55834323133351904272-if00
pin_map: arduino
restart_method: arduino
baud: 115200

[printer]
kinematics: cartesian
max_velocity: 150
max_accel: 2000
max_z_velocity: 5
max_z_accel: 30

[display]
lcd_type: uc1701
cs_pin: uno:PD7
a0_pin: uno:PD6
rst_pin: uno:PD4
spi_software_sclk_pin: uno:PB5
spi_software_miso_pin: uno:PB4
spi_software_mosi_pin: uno:PB3
contrast: 60

Is there a config related mistake I have made somewhere?
I have checked the pinout on the manufacturer's site and I verified my wiring several times, checking for correct pins and also continuity among my connections.

Its a fysetc generic 12864 LCD and the schematic shows spi miso and clock pins going to the lcd, not sure
[image]

I have used this LCD with klipper before and it worked flawlessly, post update to 0.8 I cant seem to find out what the root cause issue is.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5072961/klippy.log)

Thanks
Saurabh

saurabhceaseless, 2020-08-14:
Correction*

I noticed for uno board I had specified pin map of arduino but I was specifying hardware pin names.

I had corrected the mistake by commenting the pin map statement in my config and tried again, still no luck with the display

KevinOConnor, 2020-08-18:
As far as we know, the uc1701 code works as intended.  Generally, these types of failures are the result of an incorrect configuration, or incorrect wiring.  I recommend double checking all the wires.  Also make sure there are no unconfigured SPI devices on the same bus as the uc1701.

-Kevin

saurabhceaseless, 2020-08-18:
I agree with you that uc1701 works with klipper since I have used the exact same display with klipper on a skr1.3 before

Believe me I have checked my wiring numerous times now...I suspect the spi is not working somehow.

I have tried declaring software spi which coincides with uno’s hardware spi pins, still no luck. I’ll share with you the current display config in a susbsequent post as I’m writing this over my phone.

saurabhceaseless, 2020-08-18:
Here is the printer config

[mcu]
baud: 250000
serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
restart_method: cheetah

[mcu uno]
serial: /dev/serial/by-id/usb-Arduino__www.arduino.cc__0043_55834323133351904272-if00
#pin_map: arduino
restart_method: arduino
baud: 115200

[printer]
kinematics: cartesian
max_velocity: 150
max_accel: 2000
max_z_velocity: 5
max_z_accel: 30

[display]
lcd_type: uc1701
cs_pin: uno:PD7
a0_pin: uno:PD6
rst_pin: uno:PD4
spi_software_sclk_pin: uno:PB5
spi_software_miso_pin: uno:PB4
spi_software_mosi_pin: uno:PB3
spi_speed: 100000
#spi_bus: 1
contrast: 60
#encoder_pins: ^host:P22, ^host:P5
#click_pin: ^!uno:PD5

The rest of the pins on the arduino uno, 4 pins, are being used for controlled fans and multi pins to turn on/off stepper driver fans.

closed 2020-09-16 by github-actions[bot]

OverGlos, 2021-03-08:
I also had same problem - blank LCD display. 
Here is my solution in this case - switch the miso and mosi pin.

from
spi_software_miso_pin: PA7
spi_software_mosi_pin: PA6

to
spi_software_miso_pin: PA6
spi_software_mosi_pin: PA7