Klipper3d/klipper #3192: ZeroDivisionError in TEST_VIBRATIONS with input_shaper

antage, 2020-08-15:
Branch: `work-adxl345-20200730`
`input_shaper` is turned on:
```
[input_shaper]
shaper_freq_x: 56
shaper_freq_y: 21
shaper_type: mzv
```

Command `TEST_VIBRATIONS X=125 Y=125 Z=100 AXIS=X FREQ_START=10 FREQ_END=70 FREQ_STEP=0.5 OUTPUT=/tmp/vib-x.csv` got exception on frequency >70:
```
Traceback (most recent call last):
  File "/opt/klipper/klippy/gcode.py", line 274, in _process_commands
    handler(gcmd)
  File "/opt/klipper/klippy/gcode.py", line 140, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/opt/klipper/klippy/extras/gcode_macro.py", line 166, in cmd
    self.template.run_gcode_from_command(kwparams)
  File "/opt/klipper/klippy/extras/gcode_macro.py", line 68, in run_gcode_from_command
    self.gcode.run_script_from_command(self.render(context))
  File "/opt/klipper/klippy/gcode.py", line 290, in run_script_from_command
    self._process_commands(script.split('\n'), need_ack=False)
  File "/opt/klipper/klippy/gcode.py", line 274, in _process_commands
    handler(gcmd)
  File "/opt/klipper/klippy/gcode.py", line 140, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/opt/klipper/klippy/extras/resonance_tester.py", line 105, in cmd_TEST_VIBRATIONS
    probe_time, meas_time, meas_offset, raw_output)
  File "/opt/klipper/klippy/extras/resonance_tester.py", line 260, in _run_test
    meas_time, meas_offset)
  File "/opt/klipper/klippy/extras/resonance_tester.py", line 331, in _calc_vibrations
    self._integrate(raw_values, i, j),
  File "/opt/klipper/klippy/extras/resonance_tester.py", line 345, in _integrate
    avg_ax = integrate(1)
  File "/opt/klipper/klippy/extras/resonance_tester.py", line 344, in integrate
    return res / T
ZeroDivisionError: float division by zero
```
If I limit `FREQ_END` to 70 it works fine.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5078434/klippy.log)

KevinOConnor, 2020-08-15:
FYI, it is not useful to run TEST_VIBRATIONS with input_shaper enabled.  The input shaper code will actually suppress motor movement at the target frequency and thus will always show great results.

This suppression of movement appears to be why the divide by zero occurred (the code couldn't find the "start of the move").  This is a known issue with the testing branch.

-Kevin

dmbutyugin, 2020-09-12:
Note that with the latest developments from #3027 (and the updated branch ([docs](https://github.com/dmbutyugin/klipper/blob/work-adxl345-20200730/docs/Measuring_Resonances.md#software-installation))) this should no longer happen, because the approach to find resonances is different and the problematic code was removed.

closed 2020-09-22 by KevinOConnor