Klipper3d/klipper #3193: [Question] Fourth Axis

leonmelson, 2020-08-15:
Hi I would like to know how difficult it will be to add one more axis = A axis.

I was playing around with my own Klipper branch but I am a bit confused about were to go next.

I added [https://github.com/leonmelson/klipper/blob/master/klippy/chelper/kin_fourthaxis.c](url) file edited from kin_cartesian.c not sure if this is correct or if all xyza are required.

And I added this [https://github.com/leonmelson/klipper/blob/master/klippy/kinematics/fourthaxis.py](url) file edited from cartesian.py but I also have no clue if that was correct.

Any help will be greatly appreciated
Leon Melson

Spacemarine2018, 2020-08-17:
What is the fourth axis supposed to do?

leonmelson, 2020-08-18:
Hi Spacemarine I would like axis A to be a rotation axis around the X axis it would probably be best to just set its zero manually so no endstops needed.

The axis should move like and with XYZ with G1 so all reaches the same location at the same time if a  G1 X20 Y50 Z10 A100 is sent. I tried Manual Stepper for this with Sync=0 but the test axis would reach its end before or after the G1 command.

If Manual Stepper SYNC=0 could run in parallel a swell as be synced with the G1 ending command that could probably work but I think that might be more difficult that just adding one more axis.

It will be for a cnc so not for 3d printing #KlipperMulti lol.

Thank you in advance for you help I really appreciate it.
Leon Melson

Spacemarine2018, 2020-08-18:
I used a manual stepper to drive some additional motors, but I also didn't need a synchronization of the movements. In your case manual steppers won't work, so I don't have an easy solution.

leonmelson, 2020-08-18:
Ok and thank you for your input I am busy setting up a test system so I can try figure out what else is required and so I can get a bit better with my python coding.

KevinOConnor, 2020-08-25:
FYI, if you're looking to control the additional motor without strict synchronization then it should be possible with a manual_stepper config section.  If you want tight synchronization with moves, then it may need a bit of work - changes to gcode_move.py, toolhead.py, the chelper trapezoid motion queue, and the step generators.  It's certainly possible, but it will require a bit of work from an interested developer.

-Kevin

leonmelson, 2020-08-25:
Thank you Kevin I found some guys that use the Extruder as a 4th axis in Marlin I wounder if this will be possible for Klipper a swell?

KevinOConnor, 2020-08-25:
Yes - that's another possible solution.

-Kevin

closed 2020-09-16 by leonmelson