Klipper3d/klipper #3194: Duet2-Duex5 boards BLTouch configuration.

paulhew69, 2020-08-15:
I have a fully working RailCore II with Duet2 eth and Duex5 boards, BLTouch, runniong RRF3
After running Klipper on SKR's i thought I would swap over as I am getting to grips with the config.
Firmware done, Thermistors done, XY axis done, bit noisy at the moment, 3 Z axis motors working, just cannot get the BLTouch to work.
My RRF3 config for BL touch was this.
; Z probe and compensation definition
M558 P9 C"^zprobe.in" H5 R1 F120 T6000 A5 S0.02  ;BLTouch connected to Z probe IN pin
M950 S0 C"duex.pwm1"				             ;BLTouch Servo (S0) on duet pwm1

Which uses PWM1 on the Duex5 / E2 for the yellow wire.
White wire is plugged into the Probe in on the Duet2 ==>pin PC1
This is my config for the BLtouch under Klipper
[bltouch]
sensor_pin: PC1
control_pin: PC3
z_offset: 5

I have asked in the Klipper discord forum, people have been brill, but have not got this issue nailed.
Please could I have some help, as I am really lost.

Klippy log and cfg attached
Regards.
[printer (copy).txt](https://github.com/KevinOConnor/klipper/files/5078922/printer.copy.txt)

[klippy.log](https://github.com/KevinOConnor/klipper/files/5078908/klippy.log)

paulhew69, 2020-08-15:
It is a BLTouch 3, not a clone.
Have added
set_output_mode: 5V
still does not work.

Whole block
[bltouch]
sensor_pin: PC1
#control_pin: sx1509_duex:PIN_12 
control_pin: PC3
z_offset: 5
set_output_mode: 5V

KevinOConnor, 2020-08-15:
Unfortunately, Klipper does not support using the "sx1509" pins for the bltouch.  So, you'll need to find two pins (one for control_pin and one for sensor_pin) that are directly routed to the duet2's micro-controller.

If you do find these two pins then it should work.  Follow the docs/BLTouch.md guide to verify them.

Unfortunately, I find the duet schematics to be undecipherable, so I don't have any hints for finding two pins.

-Kevin

paulhew69, 2020-08-16:
Hi Kevin,

Appreciate the reply.

I reached out to Duet last night and surprisingly got a response from DC42, David Crocker.

This is his response
"The recommended wiring to the DueX does not use the SX1509 pins. It uses one of the PWM connectors for the control signal, e.g. PWM5 which is aliased to Heater 7. The output signal from the BLTouch is connected to one of the five endstop inputs on the main board."

I am guessing that the probe input on the Duet2 is effectively an endstop input. so, the only bit is PWM control on Heater3 needs to be looked at, how do I set it up?

I am not in your league or DC42's when it comes down to boards and code, I put things together!

I can be your test bed and try things out if that helps as I know others in the Klipper and Railcore discord's are interested in a solution and would like to use Klipper.

Kind Regards,

Paul.

KevinOConnor, 2020-08-16:
> so, the only bit is PWM control on Heater3 needs to be looked at, how do I set it up?

Klipper does not use hardware PWM for any device by default - see the FAQ: https://www.klipper3d.org/FAQ.html#do-i-have-to-wire-my-device-to-a-specific-type-of-micro-controller-pin

So, generally the only thing necessary is to obtain the "real pin name" for the pins (eg, "PA19" vs "PWM5").  After that it should work.

If you think you've found the real pin names and it still isn't working, then you could try hooking up a multimeter to the control pin, comment out the bltouch config section, add an `[output_pin test]  pin: PA19` config section, run test commands like `SET_PIN PIN=test VALUE=1`, and verify the multimeter shows the correct voltage changes.

-Kevin

paulhew69, 2020-08-16:
Hi Kevin,

Do not know if this helpful, but with more reading of the Duet info on the Duex5, it dawned on me that the heaters and PWM channels share the same logic.
Did as you suggested and created a test pin pointing to Duets' E2. I had a 24v LED plugged into E2
Reloaded the config, led was on and could not turn it off. Put a '!' infront of the pin and reloaded and led was off. issued commands and could control the led. Unplugged led. Changed my code for the BLTouch as below and it now works.

Thanks for putting up with my questions, I should of read Harder!

Here is the config if you want to put it in a prime position in the Duet-Duex Manuals.

[bltouch]
sensor_pin: PC1
control_pin: !PC3
x_offset: 2
y_offset: 42
z_offset: 5

Kind Regards,

Paul.

paulhew69, 2020-08-17:
As Kevin mentioned, Duets documentation is confusing!
Had to invert the pin using a !. All working Z_Tilt next!

closed 2020-08-17 by paulhew69