Klipper3d/klipper #3195: Delta Kinematics seperate from Z height

sentein, 2020-08-15:
I want to move forward weth a design i have in mind. To use this design i would need to be able to control the z height seperate from the x & y kinematics of the delta printer. I do not know python. Does anyone have any idea if thi is possible? Or if a z2 could be controlled seperately so i can still zero out all the towers at the start of a print. 

So what i had in mind for a procedure would be all towers home to max. Then all towers move to zero like a normal delta. Once at zero z control would be moved to z2 which would then be allowed to move the print away from the print head. The effector would move around on it set plane at zero.

Hywelmartin, 2020-08-16:
a moving bed in z?
it would be possible with manual stepper and some layer change script

sentein, 2020-08-16:
Yes a moving bed in z. I am looking to make a z belt printer. How would you disable the Delta z? Would that end up being part of the layer change script?

vladbabii, 2020-08-18:
The print can be thought of a small delta (lets say height 20 cm) and a Z-movable bed - lets say height 50 cm. If we configure the delta with only the top part, it will work fine, but we need to replace t he G0 Zx commands after homing is done.

One way this could possible be done is to override the G0-G3 commands to do different things if it's homing or not (based on a macro variable set/unset at the satart/end of the homing procedure), then use manual_stepper  to move "Z2" (the actual bed, using the SYNC parameter - https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#manual-stepper-commands

I'm not sure about how sync would affect motion, because you would need after each G0-G3 command that uses a Z motion to wait for the stepper to finish its own motion before going to the next command...

KevinOConnor, 2020-08-25:
FYI, as high-level feedback, I'm sure it is possible to implement, but an interested developer would need to take that on and implement the necessary work.

-Kevin

closed 2020-09-22 by github-actions[bot]