Klipper3d/klipper #3196: PROBE_ACCURACY - Z-Values trending towards infinity

MarvinKa, 2020-08-16:
Hello everybody,
I am using an Ender 3 with a SKR E3 Mini V1.2 Board. Also I am using an inductive probe for ABL. The rest of the hardware is stock.
When I run PROBE_ACCURACY with this setup, the reported Z-Values keep rising, while the actual Z-Position looks to be constant.
For me it seems like there must be a wrong value for the **z-step_distance**, which seems odd because I didn't change that from the default value. I just can't put all the pieces together.

Hopefully I gave you all the information you need.
Thanks in advance.

This is what PROBE_ACCURACY is reporting:
1. PROBE_ACCURACY:
Send: PROBE_ACCURACY
Recv: // PROBE_ACCURACY at X:69.500 Y:119.500 Z:10.000 (samples=10 retract=10.000 speed=5.0 lift_speed=5.0)
Recv: // probe at 69.500,119.500 is z=0.145000
Recv: // probe at 69.500,119.500 is z=0.302500
Recv: // probe at 69.500,119.500 is z=0.452500
Recv: // probe at 69.500,119.500 is z=0.692500
Recv: // probe at 69.500,119.500 is z=0.927500
Recv: // probe at 69.500,119.500 is z=1.282500
Recv: // probe at 69.500,119.500 is z=1.772500
Recv: // probe at 69.500,119.500 is z=2.140000
Recv: // probe at 69.500,119.500 is z=2.450000
Recv: // probe at 69.500,119.500 is z=2.887500
Recv: // probe accuracy results: maximum 2.887500, minimum 0.145000, range 2.742500, average 1.305250, median 1.105000, standard deviation 0.912504

2. PROBE_ACCURACY:
Send: PROBE_ACCURACY
Recv: // PROBE_ACCURACY at X:69.500 Y:119.500 Z:12.888 (samples=10 retract=10.000 speed=5.0 lift_speed=5.0)
Recv: // probe at 69.500,119.500 is z=3.385000
Recv: // probe at 69.500,119.500 is z=3.707500
Recv: // probe at 69.500,119.500 is z=4.035000
Recv: // probe at 69.500,119.500 is z=4.270000
Recv: // probe at 69.500,119.500 is z=4.487500
Recv: // probe at 69.500,119.500 is z=4.727500
Recv: // probe at 69.500,119.500 is z=5.097500
Recv: // probe at 69.500,119.500 is z=5.387500
Recv: // probe at 69.500,119.500 is z=5.637500
Recv: // probe at 69.500,119.500 is z=5.850000
Recv: // probe accuracy results: maximum 5.850000, minimum 3.385000, range 2.465000, average 4.658500, median 4.607500, standard deviation 0.787538

3. PROBE_ACCURACY:
Recv: // PROBE_ACCURACY at X:69.500 Y:119.500 Z:15.850 (samples=10 retract=10.000 speed=5.0 lift_speed=5.0)
Recv: // probe at 69.500,119.500 is z=6.037500
Recv: // probe at 69.500,119.500 is z=6.172500
Recv: // probe at 69.500,119.500 is z=6.320000
Recv: // probe at 69.500,119.500 is z=6.510000
Recv: // probe at 69.500,119.500 is z=6.877500
Recv: // probe at 69.500,119.500 is z=7.157500
Recv: // probe at 69.500,119.500 is z=7.370000
Recv: // probe at 69.500,119.500 is z=7.567500
Recv: // probe at 69.500,119.500 is z=7.785000
Recv: // probe at 69.500,119.500 is z=7.955000
Recv: // probe accuracy results: maximum 7.955000, minimum 6.037500, range 1.917500, average 6.975250, median 7.017500, standard deviation 0.658373

4. PROBE_ACCURACY:
Send: PROBE_ACCURACY
Recv: // PROBE_ACCURACY at X:69.500 Y:119.500 Z:17.955 (samples=10 retract=10.000 speed=5.0 lift_speed=5.0)
Recv: // probe at 69.500,119.500 is z=8.107500
Recv: // probe at 69.500,119.500 is z=8.232500
Recv: // probe at 69.500,119.500 is z=8.342500
Recv: // probe at 69.500,119.500 is z=8.472500
Recv: // probe at 69.500,119.500 is z=8.630000
Recv: // probe at 69.500,119.500 is z=8.870000
Recv: // probe at 69.500,119.500 is z=9.277500
Recv: // probe at 69.500,119.500 is z=9.605000
Recv: // probe at 69.500,119.500 is z=9.905000
Recv: // probe at 69.500,119.500 is z=10.092500
Recv: // probe accuracy results: maximum 10.092500, minimum 8.107500, range 1.985000, average 8.953500, median 8.750000, standard deviation 0.684395

[klippy.log](https://github.com/KevinOConnor/klipper/files/5080319/klippy.log)

BlackStump, 2020-08-16:
[If using "probe:z_virtual_endstop" then do not define a position_endstop in the stepper_z config section](https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L592)

MarvinKa, 2020-08-16:
Thanks for your reply @BlackStump, sadly that didn't fix my problem

schnello, 2020-08-17:
And the mechanical connection of Z is mechanically 100% OK?

MarvinKa, 2020-08-17:
@schnello: Yes, the coupler and the screw move at the same speed, everything is smooth and silent. Thanks for pointing the mechanical side out though.

KevinOConnor, 2020-08-18:
This looks like the stepper is losing steps on each probe halt (or perhaps the z axis has a loose grub screw).  You could try increasing the Z current, try disabling stealthchop mode, and/or try reducing probing speed.

If you want to gather further information from the software, you can run: `GET_POSITION`, `DUMP_TMC STEPPER=stepper_z`, `PROBE_ACCURACY`, `GET_POSITION`, `DUMP_TMC STEPPER=stepper_z`.  And then attach that log.

-Kevin

MarvinKa, 2020-08-18:
@KevinOConnor: the z axis is fine, no slip anywhere. I reduced the probing speed and disabled stealhchop, nothing changed. I didn't try increasing the current because I think the motor should have enough torque without stealthchop to move the axis with 2 mm/s.
Thanks for your help.

[klippy_z-dump.log](https://github.com/KevinOConnor/klipper/files/5092742/klippy_z-dump.log)

MarvinKa, 2020-08-19:
Hello everybody,
I checked every screw on the printer and as it turns out the coupling really was a little loose.
I thankful for your help but sorry for wasting your time.

closed 2020-08-19 by MarvinKa