Klipper3d/klipper #3197: Klipper LCD Menu: Change Extruder

schnello, 2020-08-16:
Hi.

I try to add a menu entry at the LCD Screen to change the active extruder:

```
[menu cube __filament __activeextruder]
type: input
input: 0
name: Ext: {printer.toolhead.extruder}
input_min: 0
input_max: 1
input_step: 1
gcode: T{1 if menu.input else 0}
```
The code above works but the name doenst show the current input. All my tries with menu.input dosnt show the wanted result.
Anyone an idea how i can add at the name the 0 and 1?

KevinOConnor, 2020-08-18:
`{1 if menu.input else 0}` does not look like valid Jinja2 template syntax to me.  I think you want something like `{% if menu.input %}1{% else %}0{% endif %}`

-Kevin

schnello, 2020-08-18:
{1 if menu.input else 0} is c&p from the actual user_menu.cfg (fan section)
and yes i have no plan how Jinja2  is to use. 

My current version is:
name: Ext: {printer.toolhead.extruder} {menu.input}
This works with small visual imperfection but it works.

Maybe @mcmatrix have an idea how to do that properly

mcmatrix, 2020-08-18:
Hi @schnello,

I don't know if it works, haven't tried it.
But maybe something like this.

As I understand the `printer.toolhead.extruder` contains active extruder name , 'extruder' or 'extruder1'.
```ini
menu cube __filament __activeextruder]
type: input
input: {['extruder', 'extruder1'].index(printer.toolhead.extruder)}
name: Ext: {menu.input}
input_min: 0
input_max: 1
input_step: 1
gcode: T{menu.input|int}
```

schnello, 2020-08-18:
>As I understand the printer.toolhead.extruder contains active extruder name , 'extruder' or 'extruder1'.

exactly

Thx i will try it after work.

schnello, 2020-08-18:
@mcmatrix 
Thank you! Works.

The final version is:
`[menu cube __filament __activeextruder]
type: input
input: {['extruder', 'extruder1'].index(printer.toolhead.extruder)}
name: Ext: extruder{menu.input|int}
input_min: 0
input_max: 1
input_step: 1
realtime: false
gcode: T{menu.input|int}`

To avoid the selection of the already active ext. (at the first press) i add also a if contion in the gcode macros. 
`[gcode_macro T0]
gcode:
 {% if (printer.toolhead.extruder) != "extruder"  %}
 ACTIVATE_EXTRUDER EXTRUDER=extruder
 {% endif %}`

`[gcode_macro T1]
gcode:
 {% if (printer.toolhead.extruder) != "extruder1"  %}
 ACTIVATE_EXTRUDER EXTRUDER=extruder1
 {% endif %}`

manu7irl, 2020-08-18:
With this you can swap extruders OTF?

schnello, 2020-08-18:
Define OTF. The macro change only the active extruder. For unload and load i have extra macros.

manu7irl, 2020-08-19:
Oh OTF = on the fly.
Like the printer is printing with say extruder1 then middle print I want to
swap to extruder2.
Will your macro do that?

schnello, 2020-08-19:
No. This entry change only the active extruder without load and unloading actions. But i dont see any problem to add the unload and load content into an extra gcode macro. Google for "M600 Klipper" should help.

Anyway. My problem is solved. Thanks a lot @mcmatrix

closed 2020-08-19 by schnello