Klipper3d/klipper #3199: Klipper new menu system : How to remove item from default menu ?

Arti4ever, 2020-08-17:
Hi @mcmatrix,

I just started to migrate from old to new menu system and I don't find how I can remove item from the default menu. 
For example, in the Main menu, how can I do to have only Octoprint and Filament sub-menus ?

With the old menu system, we could redefine the list in each menu but with the new system it seems automatic. 

Can you advise me ?
Thanks

mcmatrix, 2020-08-17:
Hi @Arti4ever,

Yes, this new menu has a different method for building the menu hierarchy.
It doesn't have manual `items` attribute anymore.
The menu hierarchy is built automatically from the config sections:
```ini
[menu __main]
...
[menu __main __control]
...
[menu __main __control __home]
...
etc
```

I think you can try to override and disable (by setting `enable: False`) top-level list items you don't want to see.
I haven't tried it.
```ini
[menu __main __control]
type: list
enable: False
name: Control
```

or you can define your own root level menu and submenus.
Something like this
```ini
[menu __mymenu]
...
[menu __mymenu __abcd1]
...
[menu __mymenu __abcd2]
...
```

and then under `display` section set custom root menu for the menu system.
`menu_root: __mymenu`

Arti4ever, 2020-08-17:
Thank you @mcmatrix ! 
First method works great.

closed 2020-08-17 by Arti4ever

th33xitus, 2020-09-25:
@mcmatrix I just stumbled across the same question and your solution of changing the menu_root made it possible to create a new, seperate menu. I just wanted to inform you that this parameter you mentioned... the `menu_root: __mymenu` is not documented anywhere. Or did i just not find it? If it's true that it isn't documented, one should consider to do so. I found it really important and especially a very simple solution to deactivate the default menu and make klipper load my custom menu.