Klipper3d/klipper #3203: How to change path to printer.cfg?

apphead, 2020-08-17:
I want to move the .cfg from /home/pi to /home/pi/klipper (in order to include it in my git). Tried to change the path in OctoKlipper plugin but that does not work. Is there an alternative way to do this?

PS: I am super happy with Klipper (coming from all sorts of other firmwares...) and love the flexibility and outstanding documentation. Now curious to see performance and quality improvements. Became my favorite firmware in a heartbeat. So thanks a ton for your efforts!

KevinOConnor, 2020-08-18:
The location of the printer config file is passed to the klippy.py program at startup.  There's a little bit of information on this in the FAQ at: https://www.klipper3d.org/FAQ.html#can-i-run-multiple-instances-of-klipper-on-the-same-host-machine

-Kevin

apphead, 2020-08-18:
For future anyone, here's what I did:
`sudo nano /etc/default/klipper`
then change 2nd parameter of KLIPPY_ARGS to whatever you like.

Thanks Kevin!

closed 2020-08-18 by apphead