Klipper3d/klipper #3205: CXY-v6-191017 3dtouch fail G28

palouf34, 2020-08-18:
Hiello
I have a 3D tronxy x5st-400 printer with the CXY-V6-191017 card.
i don't have an 'original' levelling sensor i want to use a 3dtouch.
I have connected it like this:
Black and white wire on the Y endstop.
brown and red wire on the filament2 connector
the yellow wire (servo motor) on pin 1 ( RX ) of the wifi connector

all initial tests pass, pin up and pin down.
as soon as I make a G28 the klipper goes off and the printer doesn't want to know about it.

in the octoprint log:

_Repl: G28 -> G28
Send: G28
Recv: !! BLTouch failed to deploy
Changing monitoring state from "Operational" to "Error: BLTouch failed to deploy"
Send: M112
Send: N3 M112*34
Send: N4 M104 T0 S0*37
Send: N5 M140 S0*96
Changing monitoring state from "Error: BLTouch failed to deploy" to "Offline (Error: BLTouch failed to deploy)"
Connection closed, closing down monitor_

I provide the nog and my configuration file.

thank you for your help

[printer..cfg.txt](https://github.com/KevinOConnor/klipper/files/5091555/printer.cfg.txt)
[klippy.log](https://github.com/KevinOConnor/klipper/files/5091556/klippy.log)

KevinOConnor, 2020-08-25:
Make sure you follow the OctoPrint configuration directions in the Installation document so that OctoPrint does not disconnect on an error.  Then make sure you follow the directions in the BL-Touch document to properly configure the bltouch.

https://www.klipper3d.org/Installation.html#configuring-octoprint-to-use-klipper
https://www.klipper3d.org/BLTouch.html

-Kevin

palouf34, 2020-08-25:
hello kevin,

thk’s for this help, now octoprint is ok for do not disconnet.
but my problem don’t solve.

i’m change my  pie 3 for thikclient and this error is similary.

I’m test `BLTOUCH_DEBUG COMMAND=touch_mode` and after `QUERY_PROBE` when my pin is out and this result is trigerred, I’m inverse pin in the printer.cfg and the bl-touch don’t move.

now I’m connect the servo on wifi connector on this board  because  on the filament2 is don’t work.

Have you one idea  for resolve my problem?

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/5125416/printer.cfg.txt)
[klippy.log](https://github.com/KevinOConnor/klipper/files/5125417/klippy.log)

KevinOConnor, 2020-08-28:
Unfortunately, I don't understand what you are reporting.  The best advice I can offer is to follow the direcitons in the docs/BL-Touch.md file.

-Kevin

cab404, 2020-08-28:
on unrelated note @palouf34 — where did you get this config?

palouf34, 2020-08-28:
> Unfortunately, I don't understand what you are reporting. The best advice I can offer is to follow the directions in the docs/BL-Touch.md file.
> 
> -Kevin

what I report is that despite the correct configuration, as well as the tests performed contained in the docs/BL-Touch.md file.

When I run G28 X Y is ok but as soon as I do G28 Z or simple G28, when the head arrives on the safe_home_Z then the pin comes out well but then klipper blocks with the messages contained in the log.

KevinOConnor, 2020-08-30:
It sounds like an incorrect configuration, loose wire, or faulty hardware.  Note that many bltouch devices require a pullup on the sensor pin (eg, `sensor_pin = ^!PA14` )

-Kevin

closed 2020-09-29 by github-actions[bot]