Klipper3d/klipper #3207: LCD or RPi touchscreen?

apphead, 2020-08-18:
My current setup has a 12684 display that works (apart from the stuttering encoder and reset button) plus a RPi with a 7" touchscreen. So good so far. I am wondering if I should toss the LCD and mount the RPi+display to the controller box or if I gain anything from the LCD. With Marlin and Repetier, some tasks could only be done via LCD, especially the z-babystepping. 

Thanks in advance! Big kudos to your great work. I can't stress enough how flexible klipper is. I struggled to configure my custom IDEX with RADDS DUE with Repetier (X1<->X2 Offsets were a PITA to calibrate) and decided to switch to Marlin 2.0. Then I found out about klipper and never looked back.

KevinOConnor, 2020-08-20:
Thanks.

Anything that can be done on the directly connected display can also be done via OctoPrint (or similar).  FWIW, I don't really use the lcd on my printers.  Others like them.

-Kevin

apphead, 2020-08-21:
Cheers. Byebye LCD, welcome RPi touchscreen.

closed 2020-08-21 by apphead