Klipper3d/klipper #3209: Klipper state: Shutdown for no (apparent) reason after printing fine multiple times

definitely-not-a-t-rex, 2020-08-19:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5094781/klippy.log)
Hi, yesterday I printed successfully 4 or 5 times, I turned the printer off for the night and today it doesn't want to work, I can get it to connect, see the temperatures on octoprint for a few seconds and then it says 

> Klipper state: Shutdown

I tried restarting the printer, firmware, host, using `sudo service klipper restart`, rebooting the pi, changing USB cable but it just doesn't want to work, raspbian correctly recognises the printer port ```
pi@raspberrypi:~ $ sudo ls /dev/serial/by-id/*
/dev/serial/by-id/usb-Klipper_lpc1769_1F20001722813AAFF63A6A5CC62000F5-if00
```
I don't know what else to try, it worked flawlessly yesterday.

BlackStump, 2020-08-19:
Did you look in the klippy.log? 
Also it shows a dirty version of klipper.
>Got error -1 in read: (11)Resource temporarily unavailable
Timeout with MCU 'mcu' (eventtime=1090.559135)
Transition to shutdown state: Lost communication with MCU 'mcu'

You probably need to sort out the dirty version of klipper then try again

definitely-not-a-t-rex, 2020-08-19:
> Did you look in the klippy.log?
> Also it shows a dirty version of klipper.
> Timeout with MCU 'mcu' (eventtime=1090.559135)
> Transition to shutdown state: Lost communication with MCU 'mcu'
>
> You [...]

I can't find anything on it in the docs and `make clean` followed by `make` still gives me a dirty version once flashed from sd (after renaming to ```
firmware.bin),
``` any idea?

definitely-not-a-t-rex, 2020-08-20:
> I can't find anything on it in the docs and `make clean` followed by `make` still gives me a dirty version once flashed from sd (after renaming to ` firmware.bin`),
> any idea?

removing the klipper folder fixed the dirty flag but didn't fix the issue

definitely-not-a-t-rex, 2020-08-20:
Given that it does the same with a different board too I'll open another issue with the correct log

closed 2020-08-20 by definitely-not-a-t-rex