Klipper3d/klipper #3212: menu: not possible to override or hide items on default menu

danielfmo, 2020-08-19:
Hello,

The new implementation of the menu #2344 brings two limitations regarding the previous implementation.

1. As `menu.cfg` is loaded before main configuration (lines 654 and 656 of menu.py) it is not possible to override any `menuitem` from the default `menu.cfg` #3189 .

2. There is no mechanism that allows to hide or remove an existing `menuitem` from the default `menu.cfg`

For now, in order to have an usable menu, I have all my menu configuration in my printer.cfg and commented out the line 656 from `menu.py` so that `menu.cfg` is not loaded.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5098532/klippy.log)

jakep82, 2020-08-19:
https://github.com/KevinOConnor/klipper/issues/3199

danielfmo, 2020-08-20:
Proposed solution of #3199 works

closed 2020-08-20 by danielfmo