Klipper3d/klipper #3214: Move out of range - please help

ToMiDevelop, 2020-08-20:
I had some some issues with the bed of my printer not beeing leveled. So i decided do calibrate the probe and it's offsets, measure endstops location and finaly automaticaly create a bed mesh.

The whole process went fine, but when I try to print a flat disc to test the mesh interpolation i get an error of move out of range. May someone with more experience help me? I've attached the log file, the gcode I wanted to execute and the picture of the bed mesh (I think i looks very real in comparison to printing something without any bed leveling, the high end low points seem to be measured correctly), the terminal output message and the details of the printer.

Printer: Anycubic Kossel Linear Plus, factory probe, bed, extruderm heater and other mechanical elements
Output message: Move out of range: -6.111 -4.727 285.701 [-3.000]

[image]

[klippy.log](https://github.com/KevinOConnor/klipper/files/5102353/klippy.log)
printer.cfg and the gcode file attached here have an extra txt extension added (for the github to accept them).
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/5102367/printer.cfg.txt)
[Test talerza.gcode.txt](https://github.com/KevinOConnor/klipper/files/5102371/Test.talerza.gcode.txt)

I love Klipper, it's kinematics and the way you can configure it through a simple text file, but getting bed mesh leveling to work drives me a little crazy. May I ask for help? Maybe I haven't understood something in the documentation....

Arksine, 2020-08-20:
After homing the mesh will immediately start applying adjustment.  As such, it can "slice" long moves into smaller moves, with z adjustment applied based on the mesh data.   Delta printers home to Z max and the max build height is tapered, so you can run into an issue with the first move being "sliced" with a z value that is out of range.  On your printer, max height is tapered above 255.57mm.

You have a couple of options:
1)  Move the tool straight down to Z=250 immediately after homing
2)  A more robust solution is to change your start gcode so that it clears the mesh prior to homing (with `BED_MESH_CLEAR`), homes, moves tool down to Z=250, then loads the mesh (`BED_MESH_PROFILE LOAD=default`).

ToMiDevelop, 2020-08-20:
Thank you very much. I'm going to try to use the solution with changing start gcode.

KevinOConnor, 2020-08-20:
FYI, if you've run DELTA_CALIBRATE and you've gotten a bed that tilted, it indicates something went seriously wrong somewhere.  If you attempted the DELTA_CALIBRATE with the probe, then I recommend redoing it with `DELTA_CALIBRATE METHOD=manual`.

-Kevin

ToMiDevelop, 2020-08-20:
I've changed the gcode and it helped, but I can see that the mesh calculated by the probe gives z positioning about 1mm above the bed :( I then done manual bed mesh points postioning and it got only worse, the initial layer got a weird curvature above the bed. I'm going to try manual delta calibration and hope it's going to help.

closed 2020-09-18 by github-actions[bot]

fhteagle, 2021-08-04:
Sorry to necrobump, but I have an idea that might solve this problem in a more permanent manner.

If position_min is undefined for stepper_z in printer.cfg, what about having it default to the lowest (most negative) value of the currently active bed mesh, instead of 0? 

Thanks for all the work by the devs on klipper, loving it so far, hope my idea helps this keep from coming back up every ~500 issues...