Klipper3d/klipper #3216: Implementation of Delta Keops Kinematics

ReprapRyn, 2020-08-20:
I am currently developing a new type of Delta printer, based off the Delta Keops delta robot system as seen in several research papers and manufactured commercially by Igus and Dierre Robotics

https://www.dierrerobotics.com/en/delta-robot-series/ 
[image]

https://www.igus.com/info/robotics-delta-robot

IEEE explore had an article from Dr Clavel (the original developer of linear delta technology) about the strengths of this style of linear delta robot, such as increased system rigidity and increased movement space in the X and Y plane (sacrifices some Z). It also can work as a component robot lending itself to future toolchanger and belt printer systems.
https://ieeexplore.ieee.org/document/5756938

Inverse kinematics information is in the information below from the original research paper from the University of Pamplona. also useful information from Servotronix
http://softmc.servotronix.com/wiki/DELTA_robot_-_linear_model
http://www.unipamplona.edu.co/unipamplona/portalIG/home_40/recursos/04_v19_24/revista_23/27092014/15.pdf

I would like some help with implementing the Klipper kinematics side while I build the necessary hardware. goal would be to release as a new Reprap variant.  Would the Klipper delta planner be able to support this style of linear delta robot or would I need a new C file as well?

KevinOConnor, 2020-08-20:
Interesting.

This would require both python and C host code to be written.  (It would not require new micro-controller code though.)

At a quick glance it does not seem too difficult to implement the C kinematics code.  (If the angle and position of each rail is known, then I think one should be able to calculate the rail position by rotating the desired cartesian coordinate into a coordinate system where the rail is at the origin and then use the Pythagorean formula to calculate the rail position.)

I should be able to assist with the low-level kinematic code.  If you're looking for this help, it will be necessary to have the robot built and the steppers functional.  (When ready, configure the robot using cartesian coordinates, verify stepper_buzz works on all the steppers, and then attach the resulting klippy.log file here.)  I'd also need to know the dimensions and locations of all the rails and endstops.

FWIW, I suspect the calibration of this type of robot will be harder than a typical linear delta robot.  I'd guess it will have similarities to the rotary delta (discussed a little bit at #1610).

Cheers,
-Kevin

ReprapRyn, 2020-08-20:
Got it Kevin, thanks. I should have the robot design finalized and start building it soon.

closed 2020-09-18 by github-actions[bot]