Klipper3d/klipper #3217: BLTouch pin doesn't come out for BED_MESH_CALIBRATE

LeviTheOtaku, 2020-08-20:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5104052/klippy.log)

G28 works as expected but when trying BED_MESH_CALIBRATE the pin doesn't come out with the error
 `Failed to home probe: Timeout during homing`

jakep82, 2020-08-20:
It looks like you did calibration for your bltouch z offset, but subsequent restarts show it back at 0. That needs to be fixed, and you also need to remove position_endstop from your stepper_z section.

LeviTheOtaku, 2020-08-20:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5104464/klippy.log)
The same issue happened randomly started happening again. I don't think I changed anything.

jakep82, 2020-08-20:
You still need to fix your z_offset.

LeviTheOtaku, 2020-08-20:
How do I fix it?

jakep82, 2020-08-20:
You need to run probe_calibrate, save the config when you're done, and then verify that the setting has been applied.