Klipper3d/klipper #3218: BLTouch G29 problems

Duz1nho, 2020-08-20:
Hi,

I have an ender 3.

When I execute G28 everything done ok.

But when I execute G29 , only first and second point is probed, and printer stay halted.

In the terminal the message is:

Send: G29
Recv: // probe at 62.000,25.000 is z=0.112500
Recv: // probe at 169.500,25.000 is z=0.217500
Recv: !! Move out of range: 277.000 25.000 5.000 [0.000]
Recv: !! Move out of range: 277.000 25.000 5.000 [0.000]
Recv: ok

My G29 bed mesh is configured with:

[bed_mesh]
speed: 80
horizontal_move_z: 5
mesh_min: 90,20
mesh_max: 235,235
probe_count: 3,3
fade_start: 1.0
mesh_pps: 2,2

1N4148, 2020-08-20:
Mesh min & max are referring to the probe location. Thus you've to subtract the probe offset from this values (e.g. x 235 - 45 = 190, y 235 - 10 = 225)

Duz1nho, 2020-08-21:
> 22

Solved, I do this and works well.

tks

closed 2020-08-27 by klipper-gitissuebot