Klipper3d/klipper #3219: Klipper shutting down shortly after boot on 2 different boards [Errno 16]

definitely-not-a-t-rex, 2020-08-20:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5104631/klippy.log)
Hi, I recently set up and started using klipper on my ender 5 pro using an skr 1.4 turbo but after a couple days of almost flawless prints (once calibrated pressure advance) it started not being able to boot for enough time to even home x/y and then z, looking through the log there seem to be many [Errno16] Unable to open port: `[Errno 16] could not open port /dev/serial/by-id/`, I thought it could be a board problem and since I have an skr 1.2 pro laying around I also tried it with that other board (after changing the config to one that would work for the pro and compiling the firmware for it using stm32f407 and 32kib bootloader) and it gives me the same result and that leads me to believe that the problem doesn't lie on the 1.4t but it probably depends probably on klipper. The log includes both boards with their config and all the errors. 
A typical episode of this is like this one:

```
Recv: // Klipper state: Ready
Send: M105
Recv: ok B:0.0 /0.0 T0:0.0 /0.0
Send: M105
Recv: ok B:29.7 /0.0 T0:37.4 /0.0
Send: M105
Recv: ok B:29.7 /0.0 T0:37.2 /0.0
Recv: // Klipper state: Shutdown
```
and what shows in the log after it resembles 

```
Extruder max_extrude_ratio=0.266081
Starting serial connect
got {u'count': 247, '#receive_time': 4855.599921698, u'sum': 240761, u'sumsq': 1714225, '#name': u'stats', '#sent_time': 4855.560394402}
Loaded MCU 'mcu' 77 commands (v0.8.0-721-g560d48db-20200820_193026-raspberrypi / gcc: (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] binutils: (2.31.1-11+rpi1+11) 2.31.1)
MCU 'mcu' config: BUS_PINS_i2c1=PB6,PB7 BUS_PINS_i2c2=PB10,PB11 BUS_PINS_spi3a=PC11,PC12,PC10 BUS_PINS_spi2a=PC2,PC3,PB10 STEP_DELAY=2 BUS_PINS_spi3=PB4,PB5,PB3 ADC_MAX=4095 RESERVE_PINS_crystal=PH0,PH1 CLOCK_FREQ=168000000 BUS_PINS_spi1=PA6,PA7,PA5 RESERVE_PINS_USB=PA11,PA12 MCU=stm32f407xx STATS_SUMSQ_BASE=256 BUS_PINS_spi1a=PB4,PB5,PB3 BUS_PINS_spi2=PB14,PB15,PB13
Sending MCU 'mcu' printer configuration...
Configured MCU 'mcu' (1024 moves)
Starting heater checks for heater_bed
Starting heater checks for extruder
Got error -1 in read: (11)Resource temporarily unavailable
Timeout with MCU 'mcu' (eventtime=4869.947329)
Transition to shutdown state: Lost communication with MCU 'mcu'
Dumping gcode input 3 blocks
Read 4855.584186: 'M105\n'
Read 4860.588053: 'M105\n'
Read 4865.596333: 'M105\n'
gcode state: absolute_coord=True absolute_extrude=True base_position=[0.0, 0.0, 0.0, 0.0] last_position=[0.0, 0.0, 0.0, 0.0] homing_position=[0.0, 0.0, 0.0, 0.0] speed_factor=0.0166666666667 extrude_factor=1.0 speed=25.0
```

or 

```
Extruder max_extrude_ratio=0.266081
Starting serial connect
Unable to open port: [Errno 16] could not open port /dev/serial/by-id/usb-Klipper_lpc1769_1F20001722813AAFF63A6A5CC62000F5-if00: [Errno 16] Device or resource busy: '/dev/serial/by-id/usb-Klipper_lpc1769_1F20001722813AAFF63A6A5CC62000F5-if00'
Unable to open port: [Errno 16] could not open port /dev/serial/by-id/usb-Klipper_lpc1769_1F20001722813AAFF63A6A5CC62000F5-if00: [Errno 16] Device or resource busy: '/dev/serial/by-id/usb-Klipper_lpc1769_1F20001722813AAFF63A6A5CC62000F5-if00'
Loaded MCU 'mcu' 77 commands (v0.8.0-721-g560d48db-20200820_174702-raspberrypi / gcc: (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] binutils: (2.31.1-11+rpi1+11) 2.31.1)
MCU 'mcu' config: BUS_PINS_i2c1=P0.1,P0.0 BUS_PINS_i2c0=P0.28,P0.27 BUS_PINS_i2c2=P0.11,P0.10 BUS_PINS_i2c1a=P0.20,P0.19 BUS_PINS_ssp1=P0.8,P0.9,P0.7 INITIAL_PINS=!P1.30 STEP_DELAY=2 ADC_MAX=4095 BUS_PINS_ssp0=P0.17,P0.18,P0.15 RESERVE_PINS_USB=P0.30,P0.29,P2.9 CLOCK_FREQ=120000000 MCU=lpc1769 STATS_SUMSQ_BASE=256
Sending MCU 'mcu' printer configuration...
Configured MCU 'mcu' (1024 moves)
Starting heater checks for heater_bed
Starting heater checks for extruder
Stats 1087.8: gcodein=71 mcu: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=4286 bytes_read=4622 bytes_retransmit=9 bytes_invalid=0 send_seq=221 receive_seq=206 retransmit_seq=2 srtt=0.003 rttvar=0.004 rto=0.025 ready_bytes=461 stalled_bytes=40 freq=120020670 heater_bed: target=0 temp=0.0 pwm=0.000 print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000 sysload=2.26 cputime=1.369 memavail=3338176
Stats 1088.8: gcodein=71 mcu: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=4882 bytes_read=4878 bytes_retransmit=9 bytes_invalid=0 send_seq=235 receive_seq=235 retransmit_seq=2 srtt=0.003 rttvar=0.004 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=120005348 heater_bed: target=0 temp=29.5 pwm=0.000 print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=29.4 pwm=0.000 sysload=2.26 cputime=1.382 memavail=3388168
Stats 1089.8: gcodein=76 mcu: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=4888 bytes_read=4984 bytes_retransmit=9 bytes_invalid=0 send_seq=236 receive_seq=236 retransmit_seq=2 srtt=0.002 rttvar=0.003 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=120004150 heater_bed: target=0 temp=29.5 pwm=0.000 print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=29.8 pwm=0.000 sysload=2.26 cputime=1.395 memavail=3388184
Stats 1090.8: gcodein=76 mcu: mcu_awake=0.051 mcu_task_avg=0.000059 mcu_task_stddev=0.000197 bytes_write=4903 bytes_read=5127 bytes_retransmit=9 bytes_invalid=0 send_seq=238 receive_seq=238 retransmit_seq=2 srtt=0.002 rttvar=0.003 rto=0.025 ready_bytes=7 stalled_bytes=0 freq=120002513 heater_bed: target=0 temp=29.5 pwm=0.000 print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=31.8 pwm=0.000 sysload=2.26 cputime=1.408 memavail=3388220
Stats 1091.8: gcodein=76 mcu: mcu_awake=0.051 mcu_task_avg=0.000059 mcu_task_stddev=0.000197 bytes_write=4942 bytes_read=5263 bytes_retransmit=9 bytes_invalid=0 send_seq=242 receive_seq=242 retransmit_seq=2 srtt=0.001 rttvar=0.002 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=120002079 heater_bed: target=0 temp=29.5 pwm=0.000 print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=28.9 pwm=0.000 sysload=2.26 cputime=1.422 memavail=3389092
Stats 1092.8: gcodein=76 mcu: mcu_awake=0.051 mcu_task_avg=0.000059 mcu_task_stddev=0.000197 bytes_write=4948 bytes_read=5369 bytes_retransmit=9 bytes_invalid=0 send_seq=243 receive_seq=243 retransmit_seq=2 srtt=0.001 rttvar=0.002 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=120002148 heater_bed: target=0 temp=29.4 pwm=0.000 print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=28.9 pwm=0.000 sysload=2.32 cputime=1.430 memavail=3388016
Stats 1093.8: gcodein=76 mcu: mcu_awake=0.051 mcu_task_avg=0.000059 mcu_task_stddev=0.000197 bytes_write=4970 bytes_read=5495 bytes_retransmit=9 bytes_invalid=0 send_seq=245 receive_seq=245 retransmit_seq=2 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=120001748 heater_bed: target=0 temp=29.5 pwm=0.000 print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=28.9 pwm=0.000 sysload=2.32 cputime=1.438 memavail=3413240
Stats 1094.8: gcodein=81 mcu: mcu_awake=0.051 mcu_task_avg=0.000059 mcu_task_stddev=0.000197 bytes_write=4976 bytes_read=5616 bytes_retransmit=9 bytes_invalid=0 send_seq=246 receive_seq=246 retransmit_seq=2 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=120001771 heater_bed: target=0 temp=29.5 pwm=0.000 print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=29.3 pwm=0.000 sysload=2.32 cputime=1.450 memavail=3413660
Stats 1095.8: gcodein=81 mcu: mcu_awake=0.001 mcu_task_avg=0.000008 mcu_task_stddev=0.000017 bytes_write=4998 bytes_read=5731 bytes_retransmit=9 bytes_invalid=0 send_seq=248 receive_seq=247 retransmit_seq=2 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=11 stalled_bytes=0 freq=120001699 heater_bed: target=0 temp=29.5 pwm=0.000 print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=33.1 pwm=0.000 sysload=2.32 cputime=1.463 memavail=3413840
Stats 1096.8: gcodein=81 mcu: mcu_awake=0.001 mcu_task_avg=0.000008 mcu_task_stddev=0.000017 bytes_write=5042 bytes_read=5874 bytes_retransmit=9 bytes_invalid=0 send_seq=252 receive_seq=252 retransmit_seq=2 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=120001625 heater_bed: target=0 temp=29.4 pwm=0.000 print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=29.4 pwm=0.000 sysload=2.32 cputime=1.473 memavail=3414440
[... cut ...]
Read 1079.696893: 'N2 M21*18\n'
Read 1084.632861: 'M105\n'
Read 1089.646235: 'M105\n'
Read 1094.650173: 'M105\n'
Read 1099.655427: 'M105\n'
Read 1104.661699: 'M105\n'
Read 1109.663720: 'M105\n'
Read 1114.666383: 'M105\n'
Read 1119.669592: 'M105\n'
Read 1124.672453: 'M105\n'
gcode state: absolute_coord=True absolute_extrude=True base_position=[0.0, 0.0, 0.0, 0.0] last_position=[0.0, 0.0, 0.0, 0.0] homing_position=[0.0, 0.0, 0.0, 0.0] speed_factor=0.0166666666667 extrude_factor=1.0 speed=25.0

```

jakep82, 2020-08-20:
Have you tried replacing the USB cable?

definitely-not-a-t-rex, 2020-08-20:
> Have you tried replacing the USB cable?

Both blue short ones from the boards and a longer one that I got with probably a printer or a microphone, I printed fine with the 1.4 and the cable that I used for the log just yesterday. The boards do show up using ls /dev/serial/by-id/* on all cables

BlackStump, 2020-08-20:
Have you tried a different sdcard on your Pi?

definitely-not-a-t-rex, 2020-08-20:
> Have you tried a different sdcard on your Pi?

No, do you think the could problem could lie on the sd itself or what's on it? It's the only a2 class that I have not in use elsewhere, I can try only klipper and octo on another one for a test but migrating not from a backup would take me a really long time. I can try but I don't see how it would cause `Got error -1 in read: (11)Resource temporarily unavailable` or` Unable to open port: [Errno 16] could not open port /dev/serial/by-id/usb-Klipper_lpc1769_1F20001722813AAFF63A6A5CC62000F5-if00: [Errno 16] Device or resource busy: '/dev/serial/by-id/usb-Klipper_lpc1769_1F20001722813AAFF63A6A5CC62000F5-if00'`

KevinOConnor, 2020-08-25:
FWIW, the best advice I have is in the FAQ: https://www.klipper3d.org/FAQ.html#i-keep-getting-random-lost-communication-with-mcu-errors

As far as we know, the code works as intended.  In similar reports in the past it was ultimately tracked down to loose wires, voltage fluctuations, or similar.

-Kevin

definitely-not-a-t-rex, 2020-08-26:
> FWIW, the best advice I have is in the FAQ: https://www.klipper3d.org/FAQ.html#i-keep-getting-random-lost-communication-with-mcu-errors
> 
> As far as we know, the code works as intended. In similar reports in the past it was ultimately tracked down to loose wires, voltage fluctuations, or similar.
> 
> -KevinI'

Hi, following those didn't fix it but reinstalling the os, octoprint, restoring octo backup and klipper did so I don't know what was the issue but it doesn't seem to be klipper's fault, probably some rogue process, I'll close the issue.

closed 2020-08-26 by definitely-not-a-t-rex