Klipper3d/klipper #3220: Klipper not starting after update

thatcomputerguy0101, 2020-08-20:
After updating Klipper using the non-shortcut instructions, I am getting the following log within the `klippy.log` file whenever the program restarts. I am unable to connect to Klipper through the web interface, so I was unable to issue an M112 before copying the log file. Unfortunately, I don't know what version I upgraded from other than it was likely installed sometime in late July.
```
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.8.0-721-g560d48db'
CPU: 4 core ARMv7 Processor rev 5 (v7l)
Python: '2.7.16 (default, Oct 10 2019, 22:02:15) \n[GCC 8.3.0]'
Building C code module c_helper.so
Unable to build C code module (error=256)
```
If it helps, I am running Klipper on a Raspberry Pi 2 Model B V1.1 with an image of Repetier-Server installed. The Repetier image worked before updating, so I don't think that is the problem.

thatcomputerguy0101, 2020-08-20:
Upon further exploration, using `git checkout` to go back commit 4a35d7e (from July 24, before my original installation) did not help the issue, giving the same log file minus a different Git version.

thatcomputerguy0101, 2020-08-20:
Doing a full uninstall and reinstalling Klipper from scratch corrected the problem.

closed 2020-08-20 by thatcomputerguy0101

casiomania, 2021-08-06:
Have same problem. I solve this using KIAUH, just reinstalling Klipper and Moonraker from ssh.