Klipper3d/klipper #3231: LCD Feed Menu: Not possible to feed the exact amount

schnello, 2020-08-23:
Hello.

The default feed entry is set per default to 5mm. Problem is:
With the first press 5 mm are extruded and then with the second press the value again.
@mcmatrix

mcmatrix, 2020-08-23:
@schnello Thanks for your feedback.
I'll check it and add fix if necessary.

mcmatrix, 2020-08-23:
@schnello Thank you for catching a bug.
During tomorrow I'll post a fix for it.

schnello, 2020-08-24:
@mcmatrix 
Thx. Did you also see some problems with the tune menu with the latest update? During print the tune menu is empty. Fall back to commit b490848bc9055b0e439fca88c1438d60885ffabb solves the problem

mcmatrix, 2020-08-24:
Thanks. I'll check that.
Could you attach Klipper's log regarding `Tune` menu issue?
I need to be sure that there're no errors. 

The `Tune` menu itself is selectable from the main menu
but when you enter, it's empty only back `..` ?

schnello, 2020-08-24:
@mcmatrix 
> The Tune menu itself is selectable from the main menu
> but when you enter, it's empty, only back ..?

Correct. The error is only if the printer is active. Like homing or printing.

>Could you attach Klipper's log regarding Tune menu issue?
Ok but i need some time. ATM a print is active.

schnello, 2020-08-24:
@mcmatrix 
The log:
[klippy (4).log](https://github.com/KevinOConnor/klipper/files/5120162/klippy.4.log)

mcmatrix, 2020-08-24:
@schnello Thanks.
Please update your custom menu definitions.
The commit 4c5e93d51dca2393946a67646499fef5fdd034b0 changed some printer parameters `printer.gcode -> printer.gcode_move`
Please test this PR, it should fix the input issue.

schnello, 2020-08-24:
Oh. Thanks i update my config and it works. Regarding the PR i will test it tomorrow.

schnello, 2020-08-25:
>Could you also test related PR #3236, it should fix the input issue.

I try to find the answer per google but nothing works. How can i test a unmerged PR?

mcmatrix, 2020-08-25:
Hi,

You can try this for pull request #3236:
```git
git fetch origin pull/3236/head:pr-3236
git checkout pr-3236
```
and after testing you can check out the master branch again

schnello, 2020-08-25:
Thanks. PR works fine

mcmatrix, 2020-08-25:
> Thanks. PR works fine

Thanks.

mcmatrix, 2020-08-26:
The fix was updated to the master.

closed 2020-08-26 by schnello