Klipper3d/klipper #3232: Motor died, too much current

bderancourt, 2020-08-23:
Hello,

I'm trying to make klipper working on my upgraded ender-3 (SKR mini e3 v2.0, bltouch, handpose 1.7A motors on X and Y).
It has worked for 30min for a test print before my Z motor died.

During the print I mesured these temperatures:
- X motor = 57°C
- Y motor = 35°C
- Z motor = 126°C (died after the print)
- E motor = 39°C

My tmc stepper configuration seems correct to me but during the print, I made some DUMP_TMC commands:
```
DUMP_TMC stepper_x
========== Write-only registers ==========
IHOLD_IRUN: 00081711 IHOLD=17 IRUN=23 IHOLDDELAY=8
TPWMTHRS:   00000013 TPWMTHRS=19
TPOWERDOWN: 00000014 TPOWERDOWN=20
SGTHRS:     00000000
========== Queried registers ==========
GCONF:      000001c0 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
GSTAT:      00000001 reset=1
IFCNT:      00000007 IFCNT=7
OTP_READ:   0000000c OTP_FCLKTRIM=12
IOIN:       21000040 PDN_UART=1 VERSION=0x21
FACTORY_CONF: 0000000c FCLKTRIM=12
TSTEP:      0000022d TSTEP=557
MSCNT:      000000a7 MSCNT=167
MSCURACT:   01ff0108 CUR_A=-248 CUR_B=-1
CHOPCONF:   14010053 toff=3 hstrt=5 TBL=2 MRES=4(16usteps) intpol=1
DRV_STATUS: 40170000 CS_ACTUAL=23 stealth=1
PWMCONF:    c80d0e24 PWM_OFS=36 PWM_GRAD=14 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=8 PWM_LIM=12
PWM_SCALE:  01f1006a PWM_SCALE_SUM=106 PWM_SCALE_AUTO=-15
PWM_AUTO:   00270068 PWM_OFS_AUTO=104 PWM_GRAD_AUTO=39
SG_RESULT:  000000ca SG_RESULT=202

DUMP_TMC stepper_y
========== Write-only registers ==========
IHOLD_IRUN: 00081711 IHOLD=17 IRUN=23 IHOLDDELAY=8
TPWMTHRS:   00000013 TPWMTHRS=19
TPOWERDOWN: 00000014 TPOWERDOWN=20
SGTHRS:     00000000
========== Queried registers ==========
GCONF:      000001c0 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
GSTAT:      00000001 reset=1
IFCNT:      00000007 IFCNT=7
OTP_READ:   0000000d OTP_FCLKTRIM=13
IOIN:       21000044 MS1=1 PDN_UART=1 VERSION=0x21
FACTORY_CONF: 0000000d FCLKTRIM=13
TSTEP:      00002c16 TSTEP=11286
MSCNT:      00000078 MSCNT=120
MSCURACT:   00b600a6 CUR_A=166 CUR_B=182
CHOPCONF:   14010053 toff=3 hstrt=5 TBL=2 MRES=4(16usteps) intpol=1
DRV_STATUS: 40170000 CS_ACTUAL=23 stealth=1
PWMCONF:    c80d0e24 PWM_OFS=36 PWM_GRAD=14 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=8 PWM_LIM=12
PWM_SCALE:  00000085 PWM_SCALE_SUM=133
PWM_AUTO:   000e00b5 PWM_OFS_AUTO=181 PWM_GRAD_AUTO=14
SG_RESULT:  0000001a SG_RESULT=26

DUMP_TMC stepper_z
========== Write-only registers ==========
IHOLD_IRUN: 0008120f IHOLD=15 IRUN=18 IHOLDDELAY=8
TPWMTHRS:   00000177 TPWMTHRS=375
TPOWERDOWN: 00000014 TPOWERDOWN=20
SGTHRS:     00000000
========== Queried registers ==========
GCONF:      000001c0 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
GSTAT:      00000001 reset=1
IFCNT:      00000007 IFCNT=7
OTP_READ:   0000000d OTP_FCLKTRIM=13
IOIN:       21000248 MS2=1 PDN_UART=1 DIR=1 VERSION=0x21
FACTORY_CONF: 0000000d FCLKTRIM=13
TSTEP:      000011e9 TSTEP=4585
MSCNT:      000000d8 MSCNT=216
MSCURACT:   003b00f0 CUR_A=240 CUR_B=59
CHOPCONF:   14030053 toff=3 hstrt=5 TBL=2 vsense=1 MRES=4(16usteps) intpol=1
DRV_STATUS: 40120000 CS_ACTUAL=18 stealth=1
PWMCONF:    c80d0e24 PWM_OFS=36 PWM_GRAD=14 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=8 PWM_LIM=12
PWM_SCALE:  0000001d PWM_SCALE_SUM=29
PWM_AUTO:   001b0032 PWM_OFS_AUTO=50 PWM_GRAD_AUTO=27
SG_RESULT:  0000004c SG_RESULT=76
```

But to be honnest, I don't know how to read it. Just if IRUN=18 for stepper_z is really the current sent to the motor, it explains why it died. It's just twice as much as the motor can accept.

After that, I reinstall Marlin and it works as expected, with motors heated to 30~35°.

Why don't klipper sent too much current to my motors ? Is there something to update on my config file ?
Thank you.

KevinOConnor, 2020-08-25:
> Just if IRUN=18 for stepper_z is really the current sent to the motor, it explains why it died.

An IRUN=18 (with vsense=1) corresponds to the `run_current = 0.580` you have specified in your config.  If you need a lower current then you should configure a lower current in your config.  If you don't plan to use bed_mesh (nor bed_tilt) then you may be able to configure a lower `hold_current` instead.  (If bed_mesh isn't in use then generally the Z rarely moves and one can save power by using lower amps while the Z is idle.)

If you're sure 0.580 is correct, then you may want to double check that motor power is never interrupted while Klipper is running (if motor power is interrupted it can reset the tmc2209 chip without Klipper knowing that) and make sure the driver isn't using an unusual sense resistor.

-Kevin

robthide37, 2020-08-30:
alway good idea to set the pot on the driver to a safe value encase uart fails like kevin said

closed 2020-09-29 by github-actions[bot]