Klipper3d/klipper #3233: Connection status switches every second, AttributeError

Zapfenechse, 2020-08-23:
I just updated klipper to the newest version with the KIAUH script from Github published by th33xitus.
After it I can't use the DWC 2 web interface anymore.
It only says: connection esteblished and directly after it: conncection interrupted
A tried multiple fresh installation of klipper and DWC but the problem is still there.

In the klippy.log i found the following Error:
`AttributeError: GCodeDispatch instance has no attribute 'get_status'`

But I have no clue how to fix it.
My klippy.log is attached to this issue.

I hope I'm right here with my problem and someone had a clue how to fix it.

My setup:
Ender 3 Pro
Raspberry Pi4
SKR 1.4 Board with TMC 2209

Version:
klipper: 67632105 
DWC2 for klipper: f783180e
DWC2 web UI: 3.1.1

[klippy.log](https://github.com/KevinOConnor/klipper/files/5114832/klippy.log)

BlackStump, 2020-08-23:
This is a DWC2-for Klipper Issue please report it to the [DWC2-for-Klipper Developer](https://github.com/Stephan3/dwc2-for-klipper) 

>th33xitus08/22/2020
notice as of 22-08-2020:
there were reports, that the newest commits of klipper (aug 21th and newer) will break dwc2-for-klipper and therefore it won't work anymore. if you have accidentially updated your klipper installation, roll back to commit 560d48d as this is the last known commit which works still fine. 

>to roll back, please do the following steps:
>1. cd klipper
>2. git reset --hard 560d48d
>3. sudo service klipper restart

>klipper/dwc2 should work fine again now.

Zapfenechse, 2020-08-24:
Thank you very much!

After the roll back klipper with DWC are working again

closed 2020-08-24 by Zapfenechse