Klipper3d/klipper #3234: Gcode macro broken

davidsteinberger, 2020-08-23:
I just updated to latest master (6763210). Now 1 macro that used to work fine got broken. Was there some breaking change that I missed? The macro is pretty simple:

    [gcode_macro START_PRINT]
    default_parameter_BED_TEMP: 60
    default_parameter_EXTRUDER_TEMP: 205
    gcode:
      M104 S{EXTRUDER_TEMP} ; set extruder temp
      M140 S{BED_TEMP} ; set bed temp
      M190 S{BED_TEMP} ; wait for bed temp
      G90
      G28
      G1 X5 Y10 F3000 ; move to prime
      M109 S{EXTRUDER_TEMP} ; wait for extruder temp
      G1 Z30 F1000
      M83
      G1 E-2 F5000
      M400
      { printer.gcode.action_respond_info("MSG: Clean nozzle") }
      G4 P10000
      G1 Z0.2 F1000 ; get ready to prime
      G92 E0 ; reset extrusion distance
      G1 Y160 E20 F600 ; prime nozzle
      ;G1 E-0.6 F5000
      G1 Y180 F5000 ; quick wipe
      M82
      G92 E0.0

[klippy.log](https://github.com/KevinOConnor/klipper/files/5115166/klippy.log)

BlackStump, 2020-08-23:
[Config Changes](https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Changes.md)

davidsteinberger, 2020-08-23:
@BlackStump thanks. I had the page open, but apparently haven't refreshed it. My bad.

closed 2020-08-23 by davidsteinberger