Klipper3d/klipper #3237: Failed to connect to printer

jerkwagon, 2020-08-24:
So i was printing fine last night, but noticed that i my mesh wasnt working, i realized i had 2 mesh things going on. before i had  [include meshBed.cfg] in the code but from what i gathered it was conflicting, 

This might be issue,  Right now i can connect to the octoprint, run commands from the webpage, but as soon i as i print, right after the mesh 5x5 goes through, the print stops and i get the printer is not ready error.   

I am using and ender 3 with BL touch.   not sure where i am going wrong.  Can anyone help?

Send: N18 G1 Z2.0 F3000*34
Recv: // Klipper state: Disconnect
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Changing monitoring state from "Printing" to "Error: Printer is not ready"
Send: M112
Send: N19 M112*25
Send: N20 M104 T0 S0*19
Send: N21 M140 S0*86
Changing monitoring state from "Error: Printer is not ready" to "Offline (Error: Printer is not ready)"
Connection closed, closing down monitor

[klippy.log](https://github.com/KevinOConnor/klipper/files/5120270/klippy.log)

KevinOConnor, 2020-08-25:
Make sure to follow the OctoPrint configuration instructions in the Installation document so that OctoPrint does not disconnect on an error: https://www.klipper3d.org/Installation.html#configuring-octoprint-to-use-klipper

Then, don't attempt to automatically run SAVE_CONFIG from your G29 macro.  The SAVE_CONFIG command is not intended to be used in an automated script.

-Kevin

jerkwagon, 2020-08-25:
so something new has happened, from my searches (your answers)  ran the following commands

sudo service klipper stop
make flash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
sudo service kilpper start

but now i can no longer even talk to the printer via octoprint. 
[klippy2.0.log](https://github.com/KevinOConnor/klipper/files/5121599/klippy2.0.log)

ok i followed the steps,  i also followed the steps from your link still no connection.

jerkwagon, 2020-08-25:
make flash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

only worked from the klipper folder.

jerkwagon, 2020-08-25:
not sure if this helps, contents of Makefile

 Klipper build system
#
# Copyright (C) 2016-2019  Kevin O'Connor <kevin@koconnor.net>
#
# This file may be distributed under the terms of the GNU GPLv3 license.

# Output directory
OUT=out/

# Kconfig includes
export HOSTCC             := $(CC)
export CONFIG_SHELL       := sh
export KCONFIG_AUTOHEADER := autoconf.h
export KCONFIG_CONFIG     := $(CURDIR)/.config
-include $(KCONFIG_CONFIG)

# Common command definitions
CC=$(CROSS_PREFIX)gcc
AS=$(CROSS_PREFIX)as
LD=$(CROSS_PREFIX)ld
OBJCOPY=$(CROSS_PREFIX)objcopy
OBJDUMP=$(CROSS_PREFIX)objdump
STRIP=$(CROSS_PREFIX)strip
CPP=cpp
PYTHON=python2

# Source files
src-y =
dirs-y = src

jerkwagon, 2020-08-25:
ok, so i did a clean wipe, installed, just as i had everything dialed in, i hit HOME X/Y i think it might to stopped 10 cm before the switch,  not 100% sure what caused it, so i got up to check, the ac adapter fell on the ground unplugging the pi (pi didnt fall) rebooted, and no i can no longer connect.  WTF is happening?

is there something that causes major issues from not shutting down the pi?

[3.0 klippy.log](https://github.com/KevinOConnor/klipper/files/5126373/3.0.klippy.log)

jerkwagon, 2020-08-26:
another reboot and its working, so no real fix to my issue except do a fresh install

closed 2020-08-26 by jerkwagon