Klipper3d/klipper #3238: NVIDIA Jetson Support

maverich, 2020-08-25:
Did anyone tested klipper on NVIDIA Jetson? asking to use on board CUDA processors which is not provided by any other boards

danielfmo, 2020-08-25:
To what gain? 
I can understand the curiosity and discovery of this but the truth is that an raspberry pi0 alone is powerful enough to cover the majority of uses cases (klipper wise alone)

The-Monkey-King, 2020-08-25:
Currently, I do not have one, but reading the specs, it does work with Linux. So in theory, it should work. 
However, the use cases for Jeston (aka Jetson NANO) would best suit it for virtual printing - as in running best case/testing printing simulations and not real printing. There is only so much bandwidth, including physical limiters (speed, heat, cooling, extrusion, plasticity of material), that current printing techniques can use. 

Short answer: Yes (Linux enabled) but YMMV
Use case: ?

maverich, 2020-08-25:
I guess there is some missing information about use cases, I need the CUDA power mostly for running simple deep learning models. What these deep learning models can do? Imagine a computer vision model running with camera feed detecting faults. So you dont have to stream the footage to a backend service anymore. Or a model trying to increase the speed as much as possible until it sees some side effects? when it is just simple machine learning, it works in normal raspberry pi easily but when it comes to computer vision or long sequences, then you need the power of a GPU

gui7vaz, 2020-08-25:
This is something I always imagined for 3D printers.

I work for automotive companies and we use this kind of feature to inspect
glue application in real time. Not with ML, just some customized vision
features.
(example of a specific devices we use)
https://www.quiss.net/products/rt-vision-t/
https://www.isravision.com/en/ready-to-use/bead-inspection/beadmaster3d/

KevinOConnor, 2020-08-25:
I don't know of anyone using that particular machine, but lots of people run the Klipper host code on Linux machines other than a Raspberry PI.  https://www.klipper3d.org/FAQ.html#can-i-run-klipper-on-something-other-than-a-raspberry-pi-3

-Kevin

The-Monkey-King, 2020-08-26:
Oh, ok. On this topic -- I got to see SAS (the analytics company) showcase their visual analytics programs using just a Raspberry Pi. Basically, it was a bottling line that would visually detect when the soda bottle did not match a model (ie. cap off, bottle on the side, bottle wrong color, etc). This was the Neuro [event](https://nips.cc/Expo/Conferences/2018/Schedule?demo_id=17).

Here is abstract [one](https://www.sas.com/content/dam/SAS/support/en/sas-global-forum-proceedings/2020/4432-2020.pdf) and abstract [two](https://www.sas.com/content/dam/SAS/support/en/sas-global-forum-proceedings/2020/4432-2020.pdf).

@KevinOConnor - I've been able to install and run Klipper in an Oracle VM (VirtualBox) of Debian on the Windows 10 platform. 
On my to-do list:
 1) Install in VM on Catalina MacOS (same rig as above - Ryzentosh)
 2) Install directly on Win10 platform, using Python3/C and see if I can have it run a Win10 service
 3) Same as #2 on MacOS platform

Why? Uhm,  because it's there.. and basically slimline my computing environment. Also, thinking of networking my 3D printers (USB-to-Ethernet), like Prusa does at his plant. 
 
> I guess there is some missing information about use cases, I need the CUDA power mostly for running simple deep learning models. What these deep learning models can do? Imagine a computer vision [...]

closed 2020-09-24 by github-actions[bot]