Klipper3d/klipper #3239: Best way to implement G12 nozzle cleaning macro

Lefuneste83, 2020-08-25:
Surprisingly, considering the huge amount of knowledge found on various posts on this github, I find not a single instance of someone having explained a way to implement a G12 nozzle cleaning routine. I could just write a series of fast movement at the brush position as a dumb macro, but the G12 command seems mors like a function with a few parameters as input (number of strokes, type of stroke movements...). Is there a recommended way to implement such a feature as a specific Klipper macro ? Anyone would like to share an exemple ?

theopensourcerer, 2020-08-25:
@Lefuneste83 See https://github.com/KevinOConnor/klipper/blob/master/docs/Command_Templates.md#template-expansion for a good example.

Hywelmartin, 2020-08-28:
here's a "stupid [one.. ](https://youtu.be/AoZJLQzpFMY)
```

[gcode_macro NOZZLE_CLEAN]
gcode:
	MANUAL_STEPPER STEPPER=bruch_stepper ENABLE=1 SET_POSITION=3.15 SPEED=15 ACCEL=1 MOVE=0
	MANUAL_STEPPER STEPPER=bruch_stepper ENABLE=0
	M204 S1000
	G1 X45 Y-87 Z25.5 F8000
	M204 S2000
	G1 X8 Y-59 F6000
	G1 X18 Y-59 F6000
	G1 X18 Y-69 F6000
	G1 X28 Y-69 F6000
	G1 X28 Y-79 F6000
	G1 X38 Y-79 F6000
	G1 X41 Y-82 F6000
	M204 S3000
	G1 X8 Y-59 F6000
	G1 X41 Y-82 F6000
	G1 X18 Y-59 F6000
	G1 X18 Y-69 F6000
	G1 X28 Y-69 F6000
	G1 X28 Y-79 F6000
	G1 X38 Y-79 F6000
	G1 X41 Y-82 F6000
	G1 X8 Y-59 F6000
	M204 S1000
	G1 Z40 F6000
	G1 X0 Y0 Z50 F6000
	MANUAL_STEPPER STEPPER=bruch_stepper ENABLE=1 SET_POSITION=0 SPEED=15 ACCEL=1 MOVE=3.15
```

Lefuneste83, 2020-09-19:
Here's my take on the nozzle clean macro implementation

[gcode_macro clean_nozzle]
gcode:
  SAVE_GCODE_STATE NAME=clean_nozzle_state
  G90
  G0 Z2 F5000
  G0 X116.8 Y354 F10000
  {% for wipe in range(8) %}
    {% for coordinate in [(81.8,354),(116.8,354)] %}
      G0 X{coordinate[0]} Y{coordinate[1] - 0.5 * wipe} Z0 F6000
    {% endfor %}
  {% endfor %}
  G0 X116.8 Y354 Z2 F5000
  RESTORE_GCODE_STATE NAME=clean_nozzle_state

closed 2020-10-18 by github-actions[bot]