Klipper3d/klipper #3242: Move out of range issue

WarriorClub, 2020-08-25:
I know this has been posted elsewhere in which they found a solution to this, but firstly I am not understanding it and secondly that solution does not seem to fix this I keep having.
The bed is 235 x 235 for an ender 3 bed which is in config.  at mesh _max and mesh_min below  185, 185 works, but anything above or at 190,190 it keeps giving me an out of range.  Is there something I missed? Probably but can someone point me in the right direction to this.

Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.8.0-744-g37c379dd-dirty'
CPU: 4 core ARMv7 Processor rev 3 (v7l)
Python: '2.7.16 (default, Apr  6 2019, 01:42:57) \n[GCC 8.2.0]'
Start printer at Tue Aug 25 14:17:06 2020 (1598361426.0 7.5)
===== Config file =====
[stepper_x]
step_pin = PB13
dir_pin = !PB12
enable_pin = !PB14
step_distance = .0125
endstop_pin = ^PC0
position_endstop = 0
position_max = 235
homing_speed = 50

[tmc2209 stepper_x]
uart_pin = PC11
tx_pin = PC10
uart_address = 0
microsteps = 16
run_current = 0.580
hold_current = 0.500
stealthchop_threshold = 250

[stepper_y]
step_pin = PB10
dir_pin = !PB2
enable_pin = !PB11
step_distance = .0125
endstop_pin = ^PC1
position_endstop = 0
position_max = 235
homing_speed = 50

[tmc2209 stepper_y]
uart_pin = PC11
tx_pin = PC10
uart_address = 2
microsteps = 16
run_current = 0.580
hold_current = 0.500
stealthchop_threshold = 250

[stepper_z]
step_pin = PB0
dir_pin = PC5
enable_pin = !PB1
step_distance = .0025
endstop_pin = probe:z_virtual_endstop
position_endstop = -2
position_max = 250

[tmc2209 stepper_z]
uart_pin = PC11
tx_pin = PC10
uart_address = 1
microsteps = 16
run_current = 0.580
hold_current = 0.500
stealthchop_threshold = 5

[extruder]
step_pin = PB3
dir_pin = !PB4
enable_pin = !PD2
[... 509 lines cut ...]
Stats 720.2: gcodein=112 mcu: mcu_awake=0.002 mcu_task_avg=0.000012 mcu_task_stddev=0.000043 bytes_write=17107 bytes_read=25226 bytes_retransmit=0 bytes_invalid=0 send_seq=997 receive_seq=997 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=8 freq=71997554 heater_bed: target=0 temp=25.3 pwm=0.000 print_time=721.117 buffer_time=0.000 print_stall=0 extruder: target=0 temp=25.9 pwm=0.000 sysload=0.05 cputime=5.925 memavail=3643668
Stats 721.2: gcodein=112 mcu: mcu_awake=0.005 mcu_task_avg=0.000015 mcu_task_stddev=0.000041 bytes_write=17113 bytes_read=25347 bytes_retransmit=0 bytes_invalid=0 send_seq=998 receive_seq=998 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=8 freq=71997554 heater_bed: target=0 temp=25.3 pwm=0.000 print_time=721.117 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.0 pwm=0.000 sysload=0.05 cputime=5.940 memavail=3643676
Stats 722.2: gcodein=112 mcu: mcu_awake=0.005 mcu_task_avg=0.000015 mcu_task_stddev=0.000041 bytes_write=17380 bytes_read=25548 bytes_retransmit=0 bytes_invalid=0 send_seq=1015 receive_seq=1015 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=71997552 heater_bed: target=0 temp=25.4 pwm=0.000 print_time=721.867 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.0 pwm=0.000 sysload=0.04 cputime=5.961 memavail=3643676
probe at 160.000,46.000 is z=1.682500
Stats 723.2: gcodein=112 mcu: mcu_awake=0.005 mcu_task_avg=0.000015 mcu_task_stddev=0.000041 bytes_write=17797 bytes_read=25797 bytes_retransmit=0 bytes_invalid=0 send_seq=1034 receive_seq=1034 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=57 stalled_bytes=8 freq=71997551 heater_bed: target=0 temp=25.4 pwm=0.000 print_time=724.235 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.1 pwm=0.000 sysload=0.04 cputime=5.991 memavail=3643676
Stats 724.2: gcodein=112 mcu: mcu_awake=0.005 mcu_task_avg=0.000015 mcu_task_stddev=0.000041 bytes_write=17894 bytes_read=25913 bytes_retransmit=0 bytes_invalid=0 send_seq=1037 receive_seq=1037 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=8 freq=71997551 heater_bed: target=0 temp=25.4 pwm=0.000 print_time=724.235 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.1 pwm=0.000 sysload=0.04 cputime=6.006 memavail=3643676
Stats 725.2: gcodein=112 mcu: mcu_awake=0.005 mcu_task_avg=0.000015 mcu_task_stddev=0.000041 bytes_write=18135 bytes_read=26104 bytes_retransmit=0 bytes_invalid=0 send_seq=1052 receive_seq=1052 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=71997550 heater_bed: target=0 temp=25.4 pwm=0.000 print_time=724.885 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.1 pwm=0.000 sysload=0.04 cputime=6.027 memavail=3643676
probe at 200.000,46.000 is z=1.605000
Move out of range: 240.000 46.000 5.000 [0.000]
Move out of range: 240.000 46.000 5.000 [0.000]
Stats 726.2: gcodein=112 mcu: mcu_awake=0.004 mcu_task_avg=0.000012 mcu_task_stddev=0.000032 bytes_write=18457 bytes_read=26368 bytes_retransmit=0 bytes_invalid=0 send_seq=1071 receive_seq=1071 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=32 stalled_bytes=0 freq=71997549 heater_bed: target=0 temp=25.4 pwm=0.000 print_time=726.543 buffer_time=0.862 print_stall=0 extruder: target=0 temp=26.1 pwm=0.000 sysload=0.04 cputime=6.045 memavail=3643676
Stats 727.2: gcodein=117 mcu: mcu_awake=0.004 mcu_task_avg=0.000012 mcu_task_stddev=0.000032 bytes_write=18518 bytes_read=26484 bytes_retransmit=0 bytes_invalid=0 send_seq=1074 receive_seq=1074 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=71997548 heater_bed: target=0 temp=25.3 pwm=0.000 print_time=726.543 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.1 pwm=0.000 sysload=0.04 cputime=6.059 memavail=3643676
Stats 728.2: gcodein=117 mcu: mcu_awake=0.004 mcu_task_avg=0.000012 mcu_task_stddev=0.000032 bytes_write=18557 bytes_read=26615 bytes_retransmit=0 bytes_invalid=0 send_seq=1077 receive_seq=1077 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=71997548 heater_bed: target=0 temp=25.4 pwm=0.000 print_time=726.543 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.1 pwm=0.000 sysload=0.04 cputime=6.073 memavail=3643676
Stats 729.2: gcodein=117 mcu: mcu_awake=0.004 mcu_task_avg=0.000012 mcu_task_stddev=0.000032 bytes_write=18563 bytes_read=26736 bytes_retransmit=0 bytes_invalid=0 send_seq=1078 receive_seq=1078 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=71997548 heater_bed: target=0 temp=25.4 pwm=0.000 print_time=726.543 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.1 pwm=0.000 sysload=0.04 cputime=6.085 memavail=3643676
Stats 730.2: gcodein=117 mcu: mcu_awake=0.004 mcu_task_avg=0.000012 mcu_task_stddev=0.000032 bytes_write=18569 bytes_read=26842 bytes_retransmit=0 bytes_invalid=0 send_seq=1079 receive_seq=1079 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=71997548 heater_bed: target=0 temp=25.4 pwm=0.000 print_time=726.543 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.1 pwm=0.000 sysload=0.04 cputime=6.098 memavail=3643676
Stats 731.2: gcodein=117 mcu: mcu_awake=0.003 mcu_task_avg=0.000014 mcu_task_stddev=0.000047 bytes_write=18575 bytes_read=26978 bytes_retransmit=0 bytes_invalid=0 send_seq=1080 receive_seq=1080 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=71997548 heater_bed: target=0 temp=25.4 pwm=0.000 print_time=726.543 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.0 pwm=0.000 sysload=0.04 cputime=6.110 memavail=3643676
Stats 732.2: gcodein=122 mcu: mcu_awake=0.003 mcu_task_avg=0.000014 mcu_task_stddev=0.000047 bytes_write=18581 bytes_read=27099 bytes_retransmit=0 bytes_invalid=0 send_seq=1081 receive_seq=1081 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=71997548 heater_bed: target=0 temp=25.4 pwm=0.000 print_time=726.543 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.0 pwm=0.000 sysload=0.04 cputime=6.123 memavail=3643676
Stats 733.2: gcodein=122 mcu: mcu_awake=0.003 mcu_task_avg=0.000014 mcu_task_stddev=0.000047 bytes_write=18587 bytes_read=27205 bytes_retransmit=0 bytes_invalid=0 send_seq=1082 receive_seq=1082 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=71997548 heater_bed: target=0 temp=25.4 pwm=0.000 print_time=726.543 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.0 pwm=0.000 sysload=0.04 cputime=6.136 memavail=3643676
[... cut ...] 0.03 cputime=6.207 memavail=3643676
Stats 740.2: gcodein=127 mcu: mcu_awake=0.001 mcu_task_avg=0.000008 mcu_task_stddev=0.000008 bytes_write=18629 bytes_read=28036 bytes_retransmit=0 bytes_invalid=0 send_seq=1089 receive_seq=1089 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=71997550 heater_bed: target=0 temp=25.4 pwm=0.000 print_time=726.543 buffer_time=0.000 print_stall=0 extruder: target=0 temp=26.0 pwm=0.000 sysload=0.03 cputime=6.219 memavail=3643676"

jakep82, 2020-08-25:
Please don't copy/paste your entire log.  It creates a wall of text, and prevents proper evaluation.  Attach the file per the instructions linked by the bot.

The min/max for the mesh is referencing the position of your probe ([See mesh document here](https://github.com/KevinOConnor/klipper/blob/master/docs/Bed_Mesh.md#rectangular-beds))  Your bltouch is at X -50, so with a maximum of 235 for your X axis, your maximum probe position is 235-50=185.

WarriorClub, 2020-08-25:
Thanks for that. Will do. 
That documentation helps alot. Will go back and try this.

closed 2020-09-01 by klipper-gitissuebot