Klipper3d/klipper #3248: Homing failed error, but nozzle keeps heating

ProfEngr, 2020-08-26:
This is the third time I've had sensorless homing fail for X and the nozzle has continued to heat. I don't know if it's something in my macros, Octoprint or an actual bug. Here are some relevant Octoprint log excerpts along w/ the obligatory klipper logs. https://pastebin.com/GuqcwgKy
[klippy.log](https://github.com/KevinOConnor/klipper/files/5126939/klippy.log)

KevinOConnor, 2020-08-28:
A normal homing failure is not intended to turn off the heaters.  Normally, OctoPrint would turn off the heater on an error (if printing directly from OctoPrint), or the idle_timeout would eventually turn off the heaters.

If you're having trouble with sensorless homing, the best advice I have is in the docs/Sensorless_Homing.md file.

-Kevin

ProfEngr, 2020-08-29:
I'll look at that md again. I haven't changed anything yet the X homing it failing more often. Tried changing the sensitivity up by 1 and it stops before hitting the end stop switch. Is there a homing current setting that's not listed or I missed? Or maybe it's a speed issue? 

You didn't cover why my cancel macro is being partially ignored by the firmware though. I call M112 and firmware_restart explicitly to be sure that the heaters are turned off immediately.

KevinOConnor, 2020-08-30:
There is no indication that Klipper received the `M112 and firmware_restart`.  You'll need to debug that in OctoPrint.

-Kevin

ProfEngr, 2020-08-30:
That's odd because Octoprint calls the cancel_print_pla macro for this condition. I included it in the files when I opened the issue. The macro contains those two commands. I'm seeing errors about extrusion temp in the log so the retraction commands from the beginning of the macro are being attempted.

ProfEngr, 2020-09-24:
I think I'm just going to rework my methods around what appears to be the standard behavior of the firmware.

closed 2020-09-24 by ProfEngr