Klipper3d/klipper #3249: hollow calibration cube throws error "Rescheduled timer in the past" for simplify3d, but not cura

mentaluproar, 2020-08-26:
I am calibrating my delta and part of that involves printing a simple, single walled cube.  When sliced with simplify, it get the MCU error "rescheduled timer in the past" but when I slice it in cura that same print works fine.

Why is something simplify3d sliced causing this problem?

[S3DxyzCalibration_cube.gcode.txt](https://github.com/KevinOConnor/klipper/files/5127819/S3DxyzCalibration_cube.gcode.txt)
[CURA_xyzCalibration_cube.gcode.txt](https://github.com/KevinOConnor/klipper/files/5127820/CURA_xyzCalibration_cube.gcode.txt)

mentaluproar, 2020-08-26:
EDIT: added the log.  Thanks, bot.

BlackStump, 2020-08-26:
If you are going to compare one slicer to another make sure all your speeds are the same and you do not exceed your MCU,s step rate

KevinOConnor, 2020-08-26:
From the log:
```
N28 G92 E0.0000*83
N29 G1 E-3.0000 F3600*37
```
That's a 60mm/s retraction.  With the extruder `step_distance=0.000373` it's 160858 steps per second which is greater than an atmega2560 can handle.

I don't know why cura is different from s3d.

More generally, though, I'd recommend using pressure advance (if possible) and use smaller retraction distances with slower retraction speeds.

-Kevin

mentaluproar, 2020-08-26:
I plan to, but I need to dial this in before I do anything like that. Looking at simplify again, some of these defaults are ridiculous. 

Anybody have any recommendations for a board, by the way? This is replacing a temperamental skr 1.3.

jakep82, 2020-08-26:
You can also eliminate this problem by reducing the microstep setting on your extruder.  You appear to be using a heavily geared extruder (Zesty Nimble?).  With an 8 bit MCU, you can't reliably run an extruder like that at more than 4 microsteps.

mentaluproar, 2020-08-27:
Yup, it's a nimble.  I'm running it at 16 just fine now.

mental405, 2020-09-01:
Even 16 is probably overkill for a nimble. Since you are running trinamic drivers, you can reliably set it to run in 2x or 4x microstepping with interpolation enabled. The MCU will have to process less overall steps since the step distance would be something like 0.002984 steps/mm at 2x microstepping. It should not make a difference in noise output.

mentaluproar, 2020-09-01:
I can’t even get a good first layer on this thing yet but when I do I’ll drop the microstepping rate. I could use the overhead room.

closed 2020-09-30 by github-actions[bot]