Klipper3d/klipper #3250: Cannot update menu with new items

WarriorClub, 2020-08-26:
I am trying to add a menu option to be able to do the 4 corners bed leveling.  I have made a meshbed.cfg file and placed this code below in that file.  Then included an "include meshBed" in printer.cfg.

I think some things were changed when referring to the changes documents and so I replaced all "items" with "index" as in the example config.

After restart and status the printer is ready now and everything works.  But the LCD screen has not updated and I cannot see "prepare" in order to get to these functions. 

[bed_screws]
screw1: 32,32
screw2: 203,32
screw3: 203,203
screw4: 32,203
screw5: 117,117

[menu __main]
type: list
name: Main
items:
    __tune
    __control
    __temp
    __filament
    __prepare

[menu __prepare]
type: list
enable: !toolhead.is_printing
name: Prepare
items:
    .__bed_leveling
    .__hotend_pid_tuning
    .__hotbed_pid_tuning
    .__host_restart
    .__firmware_restart

[menu __prepare __bed_leveling]
type: list
enable: !toolhead.is_printing
name: Level Corners
items:
    .__home
    .__start
    .__accept
    .__adjust
    .__abort_screws

[menu __prepare __bed_leveling __home]
type: command
name: Home All
gcode: G28
enable: !toolhead.is_printing

[menu __prepare __bed_leveling __start]
type: command
name: Start
enable: !toolhead.is_printing
gcode: BED_SCREWS_ADJUST

[menu __prepare __bed_leveling __accept]
type: command
name: Accept
enable: !toolhead.is_printing
gcode: accept

[menu __prepare __bed_leveling __adjust]
type: command
name: Adjusted
enable: !toolhead.is_printing
gcode: adjusted

[menu __prepare __bed_leveling __abort_screws]
type: command
name: Abort
enable: !toolhead.is_printing
gcode: abort

[menu __filament __feed]
type: input
name: Feed: {0:.1f}
parameter: 0
input_step: 1.0
gcode:
    M83
    G1 E{0:.1f} F100

mcmatrix, 2020-08-26:
The menu logic has been changed.
Please check `klipper/config/example-menu.cfg` for the info.
For the examples, you can also check the default menu config.
-Janar

WarriorClub, 2020-08-26:
I have checked klipper/config/example-menu.cfg but cannot see what I missed.  I followed the format of the default menu.cfg but still does not update on the lcd.  I am not sure what I am missing here.

I changed the code to
[bed_screws]
screw1: 32,32
screw2: 203,32
screw3: 203,203
screw4: 32,203
screw5: 117,117

[menu __main]
type: list
name: Main

[menu __main__prepare]
type: list
enable: !toolhead.is_printing
name: Prepare

[menu __main__prepare __bed_leveling]
type: list
enable: !toolhead.is_printing
name: Level Corners

[menu __main__prepare __bed_leveling __home]
type: command
name: Home All
gcode: G28
enable: !toolhead.is_printing

[menu __main__prepare __bed_leveling __start]
type: command
name: Start
enable: !toolhead.is_printing
gcode: BED_SCREWS_ADJUST

[menu __main__prepare __bed_leveling __accept]
type: command
name: Accept
enable: !toolhead.is_printing
gcode: accept

[menu __main__prepare __bed_leveling __adjust]
type: command
name: Adjusted
enable: !toolhead.is_printing
gcode: adjusted

[menu __main__prepare __bed_leveling __abort_screws]
type: command
name: Abort
enable: !toolhead.is_printing
gcode: abort

mcmatrix, 2020-08-26:
What I quickly noticed is the `enable` parameter.
It cannot be used this way anymore. You have to use the correct syntax (Jinja2 template for Klipper).
```py
enable: {not printer.idle_timeout.state == "Printing"}
```

There's no `Prepare` menu anymore. If you want, you can add and then change menu definitions accordingly.
In this example, I'm using the existing Setup menu. It's not tested in real printer.

```ini
[bed_screws]
screw1: 32,32
screw2: 203,32
screw3: 203,203
screw4: 32,203
screw5: 117,117

[menu __main __setup __bed_leveling]
type: list
enable: {not printer.idle_timeout.state == "Printing"}
name: Level Corners

[menu __main __setup __bed_leveling __home]
type: command
name: Home All
gcode: G28
enable: {not printer.idle_timeout.state == "Printing"}

[menu __main __setup __bed_leveling __start]
type: command
name: Start
enable: {not printer.idle_timeout.state == "Printing"}
gcode: BED_SCREWS_ADJUST

[menu __main __setup __bed_leveling __accept]
type: command
name: Accept
enable: {not printer.idle_timeout.state == "Printing"}
gcode: accept

[menu __main __setup __bed_leveling __adjust]
type: command
name: Adjusted
enable: {not printer.idle_timeout.state == "Printing"}
gcode: adjusted

[menu __main __setup __bed_leveling __abort_screws]
type: command
name: Abort
enable: {not printer.idle_timeout.state == "Printing"}
gcode: abort

```

WarriorClub, 2020-08-26:
Thank you.

I noticed a couple of lines that needed to be changed according to the example menu.cfg
{'%3d' % (menu.input*100)}%
{not printer.idle_timeout.state == "Printing"}

I replaced it with the above lines. 

But it still does not show on the LCD

[bed_screws]
screw1: 32,32
screw2: 203,32
screw3: 203,203
screw4: 32,203
screw5: 117,117

[menu __main]
type: list
name: Main: {'%3d' % (menu.input*100)}%

[menu __bed_leveling]
type: list
enable: {not printer.idle_timeout.state == "Printing"}
name: Level Corners: {'%3d' % (menu.input*100)}%

[menu __bed_leveling __home]
type: command
name: Home All: {'%3d' % (menu.input*100)}%
gcode: G28
enable: {not printer.idle_timeout.state == "Printing"}

[menu __bed_leveling __start]
type: command
name: Start: {'%3d' % (menu.input*100)}%
enable: {not printer.idle_timeout.state == "Printing"}
gcode: BED_SCREWS_ADJUST

[menu __bed_leveling __accept]
type: command
name: Accept: {'%3d' % (menu.input*100)}%
enable: {not printer.idle_timeout.state == "Printing"}
gcode: accept

[menu __bed_leveling __adjust]
type: command
name: Adjusted: {'%3d' % (menu.input*100)}%
enable: {not printer.idle_timeout.state == "Printing"}
gcode: adjusted

[menu __bed_leveling __abort_screws]
type: command
name: Abort: {'%3d' % (menu.input*100)}%
enable: {not printer.idle_timeout.state == "Printing"}
gcode: abort

mcmatrix, 2020-08-26:
Why did you add `{'%3d' % (menu.input*100)}%.`
I don't understand. Anyway, it's wrong and will not work.

I can confirm that following is working, at least the menu visuals.
```py
[menu __main __setup __bed_leveling]
type: list
index: 1
enable: {not printer.idle_timeout.state == "Printing"}
name: Level Corners

[menu __main __setup __bed_leveling __home]
type: command
name: Home All
gcode: G28
enable: {not printer.idle_timeout.state == "Printing"}

[menu __main __setup __bed_leveling __start]
type: command
name: Start
enable: {not printer.idle_timeout.state == "Printing"}
gcode: BED_SCREWS_ADJUST

[menu __main __setup __bed_leveling __accept]
type: command
name: Accept
enable: {not printer.idle_timeout.state == "Printing"}
gcode: accept

[menu __main __setup __bed_leveling __adjust]
type: command
name: Adjusted
enable: {not printer.idle_timeout.state == "Printing"}
gcode: adjusted

[menu __main __setup __bed_leveling __abort_screws]
type: command
name: Abort
enable: {not printer.idle_timeout.state == "Printing"}
gcode: abort
```
If you go to the Setup menu then your submenu is there.

mcmatrix, 2020-08-26:
Please add your Klipper log file.

WarriorClub, 2020-08-26:
Thank you! It works with the code you gave me.

WarriorClub, 2020-08-26:
I will close this ticket now.

closed 2020-09-02 by klipper-gitissuebot