Klipper3d/klipper #3251: Freeze at print

whoim2, 2020-08-26:
Hello!
https://cloud.mail.ru/public/7aJh/SFG7k7TPj

Pauses at printing.. cura 3.6 slicer, 1.4.2 octoprint, raspberry pi 3b > skr e3 mini, connect from gpio to uart via TFT port.

whoim2, 2020-08-26:
GCODE: https://dropmefiles.com/FUyeE

whoim2, 2020-08-26:
> PS: I'm just an automated script, not a human being

its true)

whoim2, 2020-08-26:
[image]

the photo shows that the defects are in the same places, that is, the nozzle stops on copies of the model in the same place

jakep82, 2020-08-26:
You're having host buffer issues as displayed on the graph from your log.  This is a known issue with Octoprint when printing something with a lot of short, fast segments.  The only solution is to use the virtual SD feature available in Klipper.

[image]

whoim2, 2020-08-26:
> You're having host buffer issues as displayed on the graph from your log. This is a known issue with Octoprint when printing something with a lot of short, fast segments. The only solution is to use the virtual SD feature available in Klipper.
> 
> [image]

Thanks. Maybe, slicer may set low quality it this moment? I use cura.

jakep82, 2020-08-26:
I can't offer slicer help because I don't use Cura.  I can only suggest that you try using virtual SD which is known to solve this issue.  

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L462

whoim2, 2020-08-26:
My friend say - use dwc2 instead octoprint.. as another way.
Tnx!

jakep82, 2020-08-26:
DWC2 and Mainsail both require virtual SD to function.  That is part of why they don't have this issue.

closed 2020-08-26 by whoim2

whoim2, 2020-09-18:
Im place fan to RPi. And.. this work!)

whoim2, 2020-09-19:
And force_turbo=1 in config.txt