Klipper3d/klipper #3252: Default Z endstop position of SKR v1.3 is not zero

gayuha, 2020-08-26:
https://github.com/KevinOConnor/klipper/blob/37c379ddc53f68808809e92a98e329f6c9a6fb2e/config/generic-bigtreetech-skr-v1.3.cfg#L33

Shouldn't it be 0? Scratched my bed pretty bad before I noticed it.

BlackStump, 2020-08-26:
It is the position of the Endstop not Z,s calibrated 0 all it was supposed to do was keep the nozzle 0.5mm above the bed so there is no possibility of scratching the bed but it does confuse a lot of folks it seems. All switch endstops can compress a further 0.5mm from trigger point to get to 0

Nearly all configs have been submitted by a user that has configured that board, the onus is always on the user to make sure a config is suitable for your usage.

when checking nozzle clearance a G1 Z0 command should issued

gayuha, 2020-08-26:
Oh I think I understand. I leveled the bed assuming it was at Z=0, and then the nozzle got lowered by 0.5 mm during printing and I couldn't figure out why.
But I think it's counter-intuitive. I assumed the endstop should stop the axis from moving any further, and here we kinda defeated the purpose by letting it go another 0.5 mm after it triggered.

jakep82, 2020-08-26:
If you follow the installation documents all the way through, you will eventually end up on the bed leveling document which explains how to calibrate your Z endstop. That should always be done before attempting a print.

https://www.github.com/KevinOConnor/klipper/tree/master/docs%2FBed_Level.md

gayuha, 2020-08-28:
Ok, I am probably just used to the way Marlin handles bed leveling. I think I'll change the endstop position to zero, seems to work as I expect. Thanks for the help!

closed 2020-08-28 by gayuha