Klipper3d/klipper #3253: klipper ignoring bed tilt and mesh leveling?

mentaluproar, 2020-08-27:
I'm having trouble getting my delta to print larger objects.  Across from tower B, things are too close to the bed.  Plastic is mashed into the bed until it just clogs the nozzle.  close to tower B, the nozzle is too far from the print bed, so the plastic won't stay down.  I've loosened the bed clips, let everything eat up, can tightened again to account for any sort of expansion but it doesnt help.

The strange part is nothing is different between no
[klippy.log](https://github.com/KevinOConnor/klipper/files/5133336/klippy.log)
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/5133337/printer.cfg.txt)

 bed leveling, mesh leveling, and bed tilt.

KevinOConnor, 2020-08-27:
The most common reason for this is an inaccurate delta calibration.  Make sure to run `DELTA_CALIBRATE METHOD=manual` and get a good level.  It's possible to use bed_mesh on top of that, but only after getting good results from DELTA_CALIBRATE.  Any mesh taken prior to a DELTA_CALIBRATE is no longer valid after running DELTA_CALIBRATE.

The vast majority of automatic probes will not work with a delta.  (Not for delta_calibrate, not for bed_mesh, and not for any other task.) Do not attempt to use an automatic probe without triple checking for a location bias first (see docs/Delta_Calibrate.md for the details).  If there is a location bias, do not use the automatic probe.

-Kevin

mentaluproar, 2020-08-27:
I just did everything again manually.  exact same result.  

I may be overlooking something.  This is a delta.  Delta calibration is the invention of the devil.  Therefore, my printer is evil.

goosst, 2020-08-29:
I also never get the result I want with the manual delta calibration, so what I do now is the following:

1. follow the delta calibration procedure
2. enable bed mesh in the printer.cfg with a low amount of points (3x3) and follow that procedure
```
[bed_mesh]
round_probe_count: 3
```
Now the bed mesh matrix is available in the printer.cfg in the `#*# [bed_mesh default]` part.

3. Then I print 3x3 bed levelling squares (similar as these https://www.thingiverse.com/thing:2789086 but then a 3x3 matrix of squares matching the probe points) and adjust the bed-mesh matrix by hand. The order in the matrix is like this:
```
#*# [bed_mesh default]
...
# points =
# 	[min_x,min_y], xxx, [max_x, min_y]
# 	xxx, xxx , xxx
# 	[min_x,max_y], xxx, [max_x, max_y]
```
4. iterate 3 untill the squares look good

KevinOConnor, 2020-08-30:
As far as I know, the code works as intended.  If you think that `DELTA_CALIBRATE METHOD=manual` is not working correctly, then you can run that command again and attach a log from the session here.  Note that you must manually clear any mesh after running DELTA_CALIBRATE, as any mesh from a previous session is no longer valid.

-Kevin

closed 2020-09-29 by github-actions[bot]