Klipper3d/klipper #3254: Skr 1.4 Turbo With 3 Wire Serial Connection to Raspberry PI4 Not connecting

girrrrrrr2, 2020-08-27:
[klippy.log.txt](https://github.com/KevinOConnor/klipper/files/5133960/klippy.log.txt)

So basically I have an skr 1.4 turbo connected to my Raspberry pi 4 Via [3 wire serial ](https://community.octoprint.org/t/simple-3-wire-serial-connection-from-pi3-to-skr-v1-3-board/10166/43) I have flashed the SKR 1.4 Turbo with the firmware.bin file (put it on the sd card and restarted it) It was setup to the right board without USB, 250000 baud (tried both smoothieware off and on multiple times)
I have enabled UART in the boot config, and disabled bluetooth, and it sees /dev/ttyAMA0
I have set the config file to look at /dev/ttyAMA0

Ive formatted the SD card for the SKR and put a new boot loader on it
Reinstalled octoprint from scratch. 

Honestly I have no clue what im doing anymore... Ive tried everything i can think of.
Oh and it doesnt work with USB either when i do ls /dev/serial/by-id/* it outputs
"ls: cannot access '/dev/serial/by-id/*': No such file or directory"

20:57:32 Connected to host via /tmp/printer @250000bps
20:57:32 Firmware version: v0.8.0-736-g2c76f752-dirty
20:57:35 Printer is not ready
The klippy host software is attempting to connect. Please
retry in a few moments.
Klipper state: Not ready
20:57:35 Printer is not ready

Honestly any help would be amazing. It is a bare skr 1.4 connected to the pi. There is nothing connected to the Skr other then the 3 wire serial. Mostly just wanted to get it connected before i start plugging things in and tidying it all up.

inks007, 2020-08-28:
If you're trying to use USB, you'll need to `make menuconfig` and enable `Use USB for communication`. It's easier, and probably better since you're using a shielded cable.

If you insist on using UART, then I would suggest you do the following:

#### Check that you've flashed your SKR properly
Flash it with USB communications enabled first. Then check that the USB device appears with `lsusb` and the USB serial device appears with `ls /dev/serial/by-id`.

#### Use UART5
You have a RasPi 4. There is no need to meddle with UART0 / UART1, serial0 / serial1 mappings and bluetooth. UART0 usually has a getty console (interferes), UART1 usually has bluetooth and uses a different UART chip that's linked to the video clock. UART0 has other quirks in the firmware.

#### Check that the RasPi UART actually works
Connect TX to RX on the RasPi (loopback) and test that you've selected the correct UART / pins / permissions. I used this:
https://github.com/cbrake/linux-serial-test

#### Reflash with serial and connect
I don't have an SKR, but the schematics show that you need to use the TFT header pins. Double check your pinout.

girrrrrrr2, 2020-08-29:
I was going to use the 3 wire serial because as i understood it helped prevent stuttering and slowdowns. But I dont believe that to be the case anymore and will use virtual SD which i have heard is the better method. So ill be trying with USB tonight ill update whether i get it working or not!

Thank you for the reply!

girrrrrrr2, 2020-08-30:
Finally got it working!

Thank you for the help!
I ended up swapping the board and the pi, and powering the pi from a usb C port... instead of via GPIO...

Not sure what part fixed it... But something did

closed 2020-09-21 by girrrrrrr2