Klipper3d/klipper #3255: Read Duet Ethernet MCU and Stepper temps.

paulhew69, 2020-08-27:
I have been through the list of Pins at the top of the Duet-Duex config file and am unable to see any reference to temps for the MCU and Steppers, as I would like to view them on MainSail.
RRF uses 'virtual heaters' 100 for the MCU temp and 101, 102 for the steppers.

DC42 replied 'The sensor is built into the microcontroller, so it doesn't use a pin', when I asked him about Logical pins.

Can these be read please?

Kind Regards,

Paul
[klippy.log](https://github.com/KevinOConnor/klipper/files/5134735/klippy.log)

paulhew69, 2020-08-27:
Someone  kindly left a further comment on my info request.
```Try something like
echo boards[0].mcuTemp.current```

Have no idea how to implement this.

KevinOConnor, 2020-08-27:
Many micro-controllers do implement a temperature sensor within the micro-controller.  However, Klipper doesn't currently support reading from these sensors.  It would require an interested developer to take that on (it will require both C micro-controller code and a little bit of Python host code).

The tmc2660 stepper motor drivers does not supply a temperature.  It only has two flags: a "very hot warning" and a "shutdown due to heat".

-Kevin

paulhew69, 2020-08-28:
OK, I cannot write C or Python, so i will not even try!

Appreciate the prompt response Kevin.

closed 2020-08-28 by paulhew69

KevinOConnor, 2020-10-07:
FYI - there is preliminary support for this up at #3407 .  I don't have a sam4e micro-controller, so I haven't been able to confirm support on that chip.  (I have tested it on my sam4s chip (Duet2 Maestro).)  If you do run a test, let me know the results (success or failure).

-Kevin