Klipper3d/klipper #3256: Python3 support? And Installation Scripts fail on Ubuntu 20.04

eduncan911, 2020-08-27:
Hello:

I'm trying to use the `install-ubuntu-18.04.sh` script to install on a full Ubuntu 20.04 OS.  The script itself doesn't seem specific to Ubuntu, except it installs a few Python 2 arm libs - which I think is an error as they aren't required.

First of all, this install script is missing any error handling and reporting.  So it makes it difficult to report errors to end users, especially to those Windows users not familiar with a bash prompt.

Also, for those wondering how I got to the commands below and the order, it's all in the `klipper/scripts/install-ubuntu-18.04.sh` shell script.  I am just showing work arounds below, and also I am skipping the env var set for the pathnames (they are at the top of the script).  If you open this shell script, you'll be able to follow along with the notes below.

The first hurdle was to get around the `ncurses` requirement.  I was not able to find a replacement lib, so I just removed the `python-virtualenv` aptitude package requirement from the script, as that doesn't exist.  I already have `python3-virtualenv` installed and fully functional with 5 Octoprints running from the same virtualenv (different config paths for them).

Next hurdle was on the setup of the virtual env for Python3.  You may see this error when trying to create a simple virtualenv via the scripts `virtualenv ${PYTHONDIR}` :

```
ModuleNotFoundError: No module named 'virtualenv.seed.via_app_data
```

I have VirtualEnvWrapper installed, which uses virtualenv but in a very clean way (not littering python crap in each virtualenv), and I exclusively use `mkvirtualenv` and `workon` for all of my virtualenvs.  I am not sure why `virtualenv` fails, as I tried the system versions as well as my virtualenvwrapper versions.  The only way around it is with:

```
python3 -m venv klippy-env
```

Now, we have a virtualenv and a new directory, `${HOME}/klippy-env`.  However, we have to manually activate it (virtualenvwrapper would normally do this for you when creating).  The script should be changed to force activation:

```
source klippy-env/bin/activate
```

We can now install the `requirements.txt` for this new virtualenv per the Klipper git source:

```
 klippy-env/bin/pip install -r klipper/scripts/klippy-requirements.txt
```

I had an error building the wheel for `cffi`.  However, it seems to have completed otherwise as running `pip install` again shows it is installed.

```
Building wheels for collected packages: cffi
  Building wheel for cffi (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/eric/klippy-env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-bkn_lox6/cffi/setup.py'"'"'; __file__='"'"'/tmp/pip-install-bkn_lox6/cffi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-b_a__cgl
       cwd: /tmp/pip-install-bkn_lox6/cffi/
  Complete output (6 lines):
  usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: setup.py --help [cmd1 cmd2 ...]
     or: setup.py --help-commands
     or: setup.py cmd --help

  error: invalid command 'bdist_wheel'
  ----------------------------------------
  ERROR: Failed building wheel for cffi
  Running setup.py clean for cffi
```

The final step was to run the `install_script()` function within this shell script, as well as it wants to start things up.  To save time, I just commented out everything else in the script and ran the last two via the script itself.

```
# Run installation steps defined above
#verify_ready
#install_packages
#create_virtualenv
install_script
start_software
```

However, the klipper systemd service seems to have failed to start:

```
$ sudo systemctl status klipper.service
● klipper.service - Starts klipper on startup
     Loaded: loaded (/etc/systemd/system/klipper.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Thu 2020-08-27 16:37:04 UTC; 52s ago
    Process: 72642 ExecStart=/home/eric/klippy-env/bin/python /home/eric/klipper/klippy/klippy.py /home/eric/printer.cfg -l /tmp>
   Main PID: 72642 (code=exited, status=1/FAILURE)

Aug 27 16:37:04 up-printer-1 systemd[1]: Started Starts klipper on startup.
Aug 27 16:37:04 up-printer-1 python[72642]: Traceback (most recent call last):
Aug 27 16:37:04 up-printer-1 python[72642]:   File "/home/eric/klipper/klippy/klippy.py", line 8, in <module>
Aug 27 16:37:04 up-printer-1 python[72642]:     import util, reactor, queuelogger, msgproto, homing
Aug 27 16:37:04 up-printer-1 python[72642]:   File "/home/eric/klipper/klippy/reactor.py", line 6, in <module>
Aug 27 16:37:04 up-printer-1 python[72642]:     import os, select, math, time, Queue as queue
Aug 27 16:37:04 up-printer-1 python[72642]: ModuleNotFoundError: No module named 'Queue'
Aug 27 16:37:04 up-printer-1 systemd[1]: klipper.service: Main process exited, code=exited, status=1/FAILURE
Aug 27 16:37:04 up-printer-1 systemd[1]: klipper.service: Failed with result 'exit-code'.
```

SO says this is an error with Python2's name of `Queue` that has renamed to `queue` for Python3...

...and i setup everything under Python3.  I thought I read where Klipper now runs under Python3.  But this error shows otherwise.

https://stackoverflow.com/questions/46363871/no-module-named-queue

---

Anyhoot... I had to start all over, but with python2.  To keep everything compatible with Klipper to continue support, i decided to continue using `virtualenv` instead of `virtualenvwrapper`.

Get used to this deprecation message:

```
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
```

On Ubuntu 20.04, python2 is installed, but only a bare bones - no pip2 is installed.  To install it, I had to pull down the old `get-pip.py` script.  Once pip2 was installed, I had to use python2 to install virtualenv:

```
python2 -m pip install virtualenv
```

Now, it's only a matter of setting up a virtualenv under python2.

```
python2 -m virtualenv klippy-env
source klippy-env/bin/activate
klippy-env/bin/pip install -r klipper/scripts/klippy-requirements.txt
```

Now, we can try to restart that existing systemd service, since it uses the same directory.

```
# remember, i already commented out all but the last two functions.  So I am only
# running the following functions within the script below:
# install_script()
# start_software()
$ klipper/scripts/install-ubuntu-18.04.sh

###### Installing system start script...

###### Launching Klipper host software...
```

Seems to be working.  let's check the status of the service:

```
$ sudo systemctl status klipper.service
● klipper.service - Starts klipper on startup
     Loaded: loaded (/etc/systemd/system/klipper.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2020-08-27 17:01:04 UTC; 19s ago
   Main PID: 73473 (python)
      Tasks: 2 (limit: 3912)
     Memory: 34.4M
     CGroup: /system.slice/klipper.service
             └─73473 /home/eric/klippy-env/bin/python /home/eric/klipper/klippy/klippy.py /ho>

Aug 27 17:01:04 up-printer-1 systemd[1]: Started Starts klipper on startup.
```

Success!

eduncan911, 2020-08-27:
Faux klipper log
[octoprint.log](https://github.com/KevinOConnor/klipper/files/5137797/octoprint.log)

eduncan911, 2020-08-27:
Faux klipper log.
[faux-klippy.log](https://github.com/KevinOConnor/klipper/files/5137798/faux-klippy.log)

inks007, 2020-08-28:
See https://github.com/KevinOConnor/klipper/issues/3097

KevinOConnor, 2020-08-30:
FWIW, best would be form someone to submit a github pull request with the desired changes to the script.

-Kevin

eduncan911, 2020-08-30:
Agreed.  During next install I'll make some tweaks...

KevinOConnor, 2020-09-22:
The virtualenv command was updated in commit 81865b59.

-Kevin

closed 2020-09-22 by KevinOConnor

boilerbots, 2021-05-23:
Would be nice to see python3 support, OctoPrint now supports python3 and is running on a bare minimum Ubuntu 20.04 install that I am playing with.