Klipper3d/klipper #3257: mid print failed

moedope, 2020-08-27:
hi all, so ive tried to do a print and it has stop mid print @ about the 10hr mark, it has happens twice and both times i resliced it to make sure it wasnt a code problem on the slicer. can you please look @ my log and shed some light on this? TY
PS. ive tried to attach a rar file as instructions say but it doesnt allow it
[klippy.log](https://github.com/KevinOConnor/klipper/files/5137827/klippy.log)

jakep82, 2020-08-27:
From your log:

```
extruder: target=240 temp=-15.5 pwm=1.000
Extrude below minimum temp
See the 'min_extrude_temp' config option for details
```

It looks your thermistor is failing or has an intermittent connection.  Before and after it dropped to -15.5 it was fluctuating quite a bit

moedope, 2020-09-01:
thanks for help, issue was thermistor wire broken

closed 2020-09-01 by moedope