Klipper3d/klipper #3263: Input Shaping for corexz

rvandermillion, 2020-08-29:
I'm running a Voron Switchwire and added input shaping to my config.  After adding [input_shaper] with more than one axis being shaped the printer fails to home.  It continues to work if you are only shaping "x" or "y" but when shaping both it fails.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5144632/klippy.log)

dmbutyugin, 2020-08-30:
@rvandermillion are you sure that if just "x" axis input shaping is enabled, your corexz setup still works? Could you double-check?

It just looks from my debugging like only "y" axis input shaping should work now on corexz, and since I do not have one, I want to double-check that I'm not the right track.

dmbutyugin, 2020-08-30:
@rvandermillion Also, could you test [this branch](https://github.com/dmbutyugin/klipper/tree/input-shaping-corexz), or simply the commit 3f2b03a18aecc5a7ccf6edc567fb9260a52b2cd0, and tell if this fixes the problem?

KevinOConnor, 2020-08-30:
FYI, the latest master branch also has the change (commit a7defdca).

-Kevin

rvandermillion, 2020-09-01:
@dmbutyugin, you are correct.  Only input shaping works when "y" is defined, "x" does not.  I will try out the branch.

dmbutyugin, 2020-09-01:
@rvandermillion By now, the fix should have already been integrated into the mainline. So you can just try to sync your install to head and see if that resolves the issue.

rvandermillion, 2020-09-01:
@dmbutyugin, I just pulled the main branch and I'm able to home with both "x" and "y" defined.  I will run the shaping routine and make sure everything still works and let you know.

rvandermillion, 2020-09-04:
Shaping worked as expected.  This issue can be closed as it has been resolved.

closed 2020-09-04 by rvandermillion