Klipper3d/klipper #3266: unusual problem, don't want z axis

iplayfast, 2020-08-29:
I'm doing an art project, where only xy coordinates are used. I've set the printer.cfg to have [stepper_z] values (on arduino), but I'm getting timeout on z axis as there isn't anything actually hooked up to it.  

I've found that just commenting out stepper_z section won't work. Is there a way to set the configuration to just assume that whatever position you want on Z is already there. (IE, ignore Z movements, Z endstops, Z homing)?

KevinOConnor, 2020-08-30:
It is necessary to define a stepper_z axis.  There is some info on disabling homing checks in the FAQ: https://www.klipper3d.org/FAQ.html#why-cant-i-move-the-stepper-before-homing-the-printer
In short, use homing_override, SET_KINEMATIC_POSITION, or define an endstop pin with a pullup and set homing_retract_dist=0.

If you don't want any heaters, do not define any heaters (nor extruders) in the config.

-Kevin

iplayfast, 2020-08-30:
Thanks, Great project btw. I don't like Patreon monthly payments, but I would like to leave a tip.  Point me to the tip jar! :)

KevinOConnor, 2020-08-30:
> Thanks, Great project btw. I don't like Patreon monthly payments, but I would like to leave a tip. Point me to the tip jar! :)

Thanks.  I understand, but at this time I only have Patreon.  In any case, I appreciate the feedback.

Cheers,
-Kevin

iplayfast, 2020-09-02:
I'm sorry, I must be being dense. 
I set it up as 
[code]
[homing_override]
SET_KINEMATIC_POSITION: Z=0
endstop_pin: ^!PB3
[/code]

and it came back with 
Recv: // No option 'gcode' in section: 'homing_override'
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Recv: // Printer is halted

what gcode is it expecting?
and is my syntax correct?

KevinOConnor, 2020-09-03:
You'll need to attach the Klipper log in order for us to provide any meaningful assistance.  That said, you'll also need to read through the documentation in example-extras.cfg for how one configures a [homing_override] config section.

-Kevin

iplayfast, 2020-09-13:
Sorry, I'm only able to work on this project on the weekends, and sometimes not even then. Didn't mean to ghost you!
[klippy.log](https://github.com/KevinOConnor/klipper/files/5214962/klippy.log)
So I've carefully gone through everything, and have to to the point where doing a status gives no errors or warnings.
When I try to print, it comes back with 
**Send: N164 G1 X-69.538 Y-13.201 E14.95086*86
Recv: !! Must home axis first: -69.538 -13.201 0.350 [14.951]
WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error: Must home axis first: -69.538 -13.201 0.350 [14.951]**

However when I send G28 
Send: G28
Recv: ok
Send: M105
It has no complaint.

Again to review, I only want x/Y axis, and would like to ignore any commands related to Z or extruder, or heat, or fans.

KevinOConnor, 2020-09-22:
If you specify a homing_override section, the gcode in that section _must_ home every axis - see the description in config/example-extras.cfg .

-Kevin

iplayfast, 2020-09-27:
Thanks that solved that problem, and I was able to resolve extrusion problems at prusaslicer by using machine-kit gcode flavor.
It now prints the outline of the model (not really prints since there is no extruder) but when it start doing infile it starts being out of range. 

I have no idea why the range is suddenly jumping. 
The serial log near where the error is, from octo print shows : 

```
020-09-27 22:37:12,278 - Send: N152 G1 X294.292 Y326.575 A49.37192*85
2020-09-27 22:37:12,595 - Recv: ok
2020-09-27 22:37:12,619 - Send: N153 G1 X289.947 Y327.893 A49.59978*91
2020-09-27 22:37:12,631 - Recv: ok
2020-09-27 22:37:12,644 - Send: N154 G1 X289.741 Y328.281 F7800.000*93
2020-09-27 22:37:12,655 - Recv: ok
2020-09-27 22:37:12,667 - Send: N155 G1 A47.59978 F2400.00000*33
2020-09-27 22:37:12,677 - Recv: ok
2020-09-27 22:37:12,690 - Send: N156 G92 A0*65
2020-09-27 22:37:12,699 - Recv: ok
2020-09-27 22:37:12,713 - Send: N157 G1 X228.140 Y259.168 F7800.000*81
2020-09-27 22:37:12,727 - Recv: !! Move out of range: 735.728 810.643 1.050 [0.000]
2020-09-27 22:37:12,743 - WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error: Move out of range: 735.728 810.643 1.050 [0.000]
2020-09-27 22:37:12,751 - Recv: ok
2020-09-27 22:37:12,762 - Send: N158 G1 A2.00000 F2400.00000*23
2020-09-27 22:37:12,772 - Recv: ok
2020-09-27 22:37:12,786 - Send: N159 G1 F1800.000*84
2020-09-27 22:37:12,796 - Recv: ok
2020-09-27 22:37:12,811 - Send: N160 G1 X226.400 Y258.732 A2.08998*104
2020-09-27 22:37:12,824 - Recv: !! Move out of range: 733.988 810.207 1.050 [0.000]
2020-09-27 22:37:12,839 - WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error: Move out of range: 733.988 810.207 1.050 [0.000]

```
which as far as I understand looks ok. (I don't understand the * things, but that's throughout so probably not it).

[serial.log.zip](https://github.com/KevinOConnor/klipper/files/5289143/serial.log.zip)
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/5289144/klippy.log.zip)
[fromocto.gcode.zip](https://github.com/KevinOConnor/klipper/files/5289145/fromocto.gcode.zip)

iplayfast, 2020-09-27:
>Thanks that solved that problem, and I was able to resolve extrusion problems at prusaslicer by using machine-kit gcode flavor. It now prints the outline of the model (not really prints since there is no extruder) but when it start doing infile it starts being out of range.
 
 I have no idea why the position is suddenly jumping.
There is a G92 code there but it shouldn't affect the x or y positions. (should it?)
 The serial log near where the error is, from octo print shows :
 
 ```
 020-09-27 22:37:12,278 - Send: N152 G1 X294.292 Y326.575 A49.37192*85
 2020-09-27 22:37:12,595 - Recv: ok
 2020-09-27 22:37:12,619 - Send: N153 G1 X289.947 Y327.893 A49.59978*91
 2020-09-27 22:37:12,631 - Recv: ok
 2020-09-27 22:37:12,644 - Send: N154 G1 X289.741 Y328.281 F7800.000*93
 2020-09-27 22:37:12,655 - Recv: ok
 2020-09-27 22:37:12,667 - Send: N155 G1 A47.59978 F2400.00000*33
 2020-09-27 22:37:12,677 - Recv: ok
 2020-09-27 22:37:12,690 - Send: N156 G92 A0*65
 2020-09-27 22:37:12,699 - Recv: ok
 2020-09-27 22:37:12,713 - Send: N157 G1 X228.140 Y259.168 F7800.000*81
 2020-09-27 22:37:12,727 - Recv: !! Move out of range: 735.728 810.643 1.050 [0.000]
 2020-09-27 22:37:12,743 - WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error: Move out of range: 735.728 810.643 1.050 [0.000]
 2020-09-27 22:37:12,751 - Recv: ok
 2020-09-27 22:37:12,762 - Send: N158 G1 A2.00000 F2400.00000*23
 2020-09-27 22:37:12,772 - Recv: ok
 2020-09-27 22:37:12,786 - Send: N159 G1 F1800.000*84
 2020-09-27 22:37:12,796 - Recv: ok
 2020-09-27 22:37:12,811 - Send: N160 G1 X226.400 Y258.732 A2.08998*104
 2020-09-27 22:37:12,824 - Recv: !! Move out of range: 733.988 810.207 1.050 [0.000]
 2020-09-27 22:37:12,839 - WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error: Move out of range: 733.988 810.207 1.050 [0.000]
 ```
 
 which as far as I understand looks ok. (I don't understand the * things, but that's throughout so probably not it).
 
 [serial.log.zip](https://github.com/KevinOConnor/klipper/files/5289143/serial.log.zip)
 [klippy.log.zip](https://github.com/KevinOConnor/klipper/files/5289144/klippy.log.zip)
 [fromocto.gcode.zip](https://github.com/KevinOConnor/klipper/files/5289145/fromocto.gcode.zip)

iplayfast, 2020-09-27:
I tried manually editing out the G92 codes, and she's working as she should.
Is there a way of having klipper ignore these codes for A0?

KevinOConnor, 2020-10-03:
Best would probably be to filter them out of your gcode file using some kind of tool.

It may be possible to implement a gcode_macro that overrides G92 and then ignores some of these commands.  However, that approach is complex and not something I'd be able to assist with.

-Kevin

iplayfast, 2020-10-05:
Thanks for your help.

closed 2020-10-05 by iplayfast