Klipper3d/klipper #3267: Internal error on command:"G28"

jalanjarosz, 2020-08-30:
Changed from S-Curve back to Klipper Main and updated now that S-Curve has been merged.

Attempted to simply home printer, which has been working well for about a year on Klipper and received the Internal error on command:"G28" issue.

Re-Flashed both MEGA controllers, attempted the G28 Home command and had same issue.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5146184/klippy.log)

dmbutyugin, 2020-08-30:
@jalanjarosz It appears that your config has invalid values
```
[input_shaper]
shaper_freq_x = 0.0194
shaper_freq_y = 0.0141
```
Namely, it looks like the values were not updated from the old format - seconds - to the new format in Hz.

jalanjarosz, 2020-08-30:
@dmbutyugin That was it.   Thanks!

Is there a quick reference for and adjustment to the input_shaper values, from old to new, or should I go though the print/measure process again?

(on a side note, this made me upgrade my SBC a little faster)  

Thanks!

jalanjarosz, 2020-08-30:
Decided to re-run the ringing tests and tests are coming out really nice. 
Closing ticket as issue is resolved.
Thank you @dmbutyugin

closed 2020-08-30 by jalanjarosz