Klipper3d/klipper #3273: Calibrating Probe Z Offset

quad121, 2020-09-01:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5158371/klippy.log)

I am having difficulty saving 'Tuned' Z Offset settings.
I was advised to run the 'PROBE_CALIBRATE' command which after the printer performs a probe and shows a measurement on screen, i then type 'ACCEPT', which states that 'Manual probe failed! Use TESTZ commands to possition the nozzle prior to running ACCEPT.
I tried then to run the TESTZ Z=*.** (* donotes numerals) which then i receive 'Unknown Command'?

So at the moment i am unable to save a new Z Offset so having to set this manually at the beginning of every print.

Is there any other way to save the settings? I've tried changing them in the 'printer.cfg' file under BLTouch but it doesn't do anything.

Any help would be much appreciated.

jakep82, 2020-09-01:
After running probe_calibrate, you need to follow the instructions on the [Bed Level document](https://github.com/KevinOConnor/klipper/blob/master/docs/Bed_Level.md#the-paper-test) for the paper test.  Before choosing accept, you should use testz to move the nozzle down until it's just above the bed.

quad121, 2020-09-02:
Ok, I Followed the instruction on the 'Paper Test' section, For some unknown reason the printer accepted the 'TESTZ' command and I have now managed to do the  'Probe Clibration'.
I don't know if it was related? , but previous to retrying this procedure I did change the axis of X and Y as they where working 'mirrored'. I was homing at 0,0 rather than at max point!
Anyway this issue has now been resolved. Many Thanks

quad121, 2020-09-02:
This issue has been resolved!

closed 2020-09-02 by quad121