Klipper3d/klipper #3275: Tuning Tower Size

supersebbo, 2020-09-03:
Is there any reason the pressure advance tuning tower needs to be as large as it is? 
https://github.com/KevinOConnor/klipper/blob/master/docs/prints/square_tower.stl

With the suggestion that tuning needs to be completed for each type, colour and even batch of filament, the tower consumes a large amount of raw material.   

Could the sides of the tower be reduced?  Wall thickness reduced?  Would this negatively affect the test output?

KevinOConnor, 2020-09-06:
I'm sure there are many ways to tune pressure advance.  What we have now works well for many users.

The wall lengths and wall thinkness were chosen to emphasize the effects of pressure.

If you find there is too much plastic being used, I recommend choosing a START and FACTOR setting in the TUNING_TOWER command that allows you to reliably find the ideal pressure setting within the first 1cm of the print.  You can cancel the print after that.

-Kevin

ProfEngr, 2020-09-16:
Instead of cancelling, you could also just drop the majority of the part below the bed in your slicer. Leave 1-2cm above if you're fine tuning the setting. Some slicers offer the option to stop at a certain height or split the part and delete the top section.

closed 2020-10-15 by github-actions[bot]