Klipper3d/klipper #3276: Unable to Update Z Offset Value!

quad121, 2020-09-03:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5168811/klippy.log)

My problem is after changing the Nozzle it throw the Z-Offset out, so I made the'Tune' as it was printing the first layer and noted the figure. The nozzle is too low, I deducted this figure from the 'BLTouch' section in the 'SAVE_CONFIG' section and updated that figure. I then did a 'FIRMWARE_RESTART' Tried  to print again but no difference! I then tried performing the 'Probe_Calibrate' again and used the paper under the nozzle and adjusted the nozzle down using the 'TESTZ' commands, when i had this right i 'ACCEPT' . This still doesnt seem to save nor change the Offset. Any help would be most appreciated!

quad121, 2020-09-04:
I have also tried changing the value of the z-offset in the 'printer' firmware folder under 'bltouch', but it doesn't have any effect.

KevinOConnor, 2020-09-11:
The log doesn't show that SAVE_CONFIG was ever run.  You need to type that command into the OctoPrint terminal.  If you continue to have issues, issue an M112 command immediately after running SAVE_CONFIG and provide that log file.

-Kevin

closed 2020-10-11 by github-actions[bot]