Klipper3d/klipper #3277: Heater extruder_heater not heating at expected rate

Yashchoor, 2020-09-03:
My Anycubic Kossel Linear Plus randomly stops printing, throwing this error. Oddly enough, the printer stops responding to any control over the LCD, always hanging up on 176 degrees Celsius, "heating" to 200 degrees. What could be the issue? Heater cables seem fine. After issuing FIRMWARE_RESTART and then trying to heat up the extruder via LCD the machine does not turn the heater on (extruder fan spins all right), however I can control the bed heater just fine with the LCD. I also noticed, that after restarting the firmware I cannot heat up either of the heaters via OctoPrint.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5168890/klippy.log)

jakep82, 2020-09-03:
It looks like you have a hardware issue.  After your hotend has been at 200 for a while, it begins cooling down to 173 before Klipper shuts down for safety reasons.  Klipper is sending full power (pwm=1.000) during this time.  

Your temperatures look stable so it's likely not the thermistor.  Check to see if you have a damaged wire going to your heater cartridge, and also make sure the mosfet controlling your hotend is working properly.

Yashchoor, 2020-09-04:
Upon checking the heater wires continuity as @jakep82 suggested, I found out that although the cables looked just fine, their cores were broken, when bent in certain way, there was no contact.

closed 2020-09-04 by Yashchoor