Klipper3d/klipper #3279: Idea: vary speed to avoid resonance

bob-u, 2020-09-04:
I've been watching video below and I'm wondering if avoiding constant speeds, by accelerating and decelarating continuously, could decrease resonance at higher travel rates?

https://www.youtube.com/watch?time_continue=276&v=dXR2wDDFFm4&feature=emb_title

Hairyloon, 2020-09-04:
Interesting idea.
The thing about resonance is that it happens at specific frequencies, so in theory all one would need to do is determine those frequencies for that printer and avoid them.
Possibly your way may be easier.

KevinOConnor, 2020-09-06:
There was a lengthy discussion on resonance in issue #57 .  Ultimately, we decided to pursue "input shaping": https://www.klipper3d.org/Resonance_Compensation.html

-Kevin

dmbutyugin, 2020-09-08:
To add some extra perspective: constant velocity movements rarely cause vibrations in 3D printer. There are only a few occasions when this can happen (like if the belts are too tight causing a pattern on the print), and these should be fixed differently. Acceleration, deceleration and sudden print direction changes are the usual causes of ringing - and then the toolhead does not move at a constant velocity.

closed 2020-10-01 by bob-u