Klipper3d/klipper #3283: endstops not triggering with skr pro 1.2

nagamotoK, 2020-09-05:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5177795/klippy.log)
I tried to upgrade a _sidewinder x1_ with an _skr pro 1.2_ and klipper. Due to the board change I don't have any marlin values to compare to or can even be sure that the problem is not due to bad wiring.  

At the moment the endstops wont trigger. 

I tried pullup ^, pulldown ~, inverted ! and I always get 
> Send: QUERY_ENDSTOPS
> Recv: x:open y:open z:open

even when I move the axis manually to touch the endstop.
Thanks for any help! 

### Board layout:
[image]
### wiring:
[image]

BlackStump, 2020-09-05:
[have you read the skr pro manual](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/SKR-PRO-V1.2/manual/SKR-PRO-V1.2-user%20manual.pdf)

Pay attention to what it says about the TMC2209

nagamotoK, 2020-09-05:
ah thanks - I seen it but I didn't understood it at the moment. Would you use stallguard or endstops?

for the com. loss: I think this works now. I taped up the two outer usb pins in the usb type A plug, see here: https://www.klipper3d.org/FAQ.html#i-keep-getting-random-lost-communication-with-mcu-errors

nagamotoK, 2020-09-05:
I bend the stall guard pin out of the way, it is not connecting or touching anything. 

Send: QUERY_ENDSTOPS gives now:

> Recv: x:TRIGGERED y:TRIGGERED z:TRIGGERED

but still moving the eg. the x axis to the endstops does not result x:open 

adding "^" to the config seems to have no effect, "!" reverses to open  but no change

nagamotoK, 2020-09-05:
I might be on to something the mks gen l does provide 5V to the sensor and the skr pro just 3.3V. Is this the reason why the sensor does not work anymore? 

[image]

nagamotoK, 2020-09-05:
ok -- this was the problem, the inductive endstops form the sidewinder require 5V, I used the first pin from extension 2 for this

closed 2020-09-05 by nagamotoK