Klipper3d/klipper #3286: No next step, I did not found any connection with other simalar issues

mistrjirka, 2020-09-06:
Hello, I am running klipper on debian Core 2 duo with mechanical HDD. When I am printing very slowly I am getting error mentioned bellow. I read these issues https://github.com/KevinOConnor/klipper/issues/2112 this did not help I was printing slowly changing klipper priority did not help and I do not have slow computer for klipper, https://github.com/KevinOConnor/klipper/issues/1390 there is solution some bug fix, but I have newly installed klipper. https://github.com/KevinOConnor/klipper/issues/1012 I have no idea what even is Force_move and it is nowhere to be found in my gcode. 
How I said I am printing very slowly 5mm/s. 
I attached log, previously I found in the file the error and copied it here. The whole log has 51MB.
[printer.txt](https://github.com/KevinOConnor/klipper/files/5179250/printer.txt)

  
[2klippy.log.zip](https://github.com/KevinOConnor/klipper/files/5179252/2klippy.log.zip)

Read 195509.617502: 'N63527 G1 X111.605 Y110.039 E0.00746*106\n'
Read 195509.619468: 'N63528 G1 E-0.10000 F4800.000*49\n'
Read 195509.621273: 'N63529 M400*44\n'
gcode state: absolute_coord=True absolute_extrude=False base_position=[0.0, 0.0$
MCU 'mcu' shutdown: No next step
This is generally indicative of an intermittent
communication failure between micro-controller and host.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

KevinOConnor, 2020-09-11:
Can you provide the gcode file that was printing at the time of the error?

-Kevin

mistrjirka, 2020-09-18:
https://pastebin.pl/view/984d4359 Sorry for delay. I was not sure which one it was. Btw it does not matter if it is Prusaslicer or Cura both create same error. This one failed in like 70% printing

mistrjirka, 2020-09-22:
New update. Printed again on slow settings and about 20% into the print it stopped with same error. Here are logs from octoprint klippy and config
[bugrepport.zip](https://github.com/KevinOConnor/klipper/files/5263405/bugrepport.zip)

mistrjirka, 2020-09-22:
[image]
[image]
[image]
[image]
[image]

KevinOConnor, 2020-09-22:
Are you running a 32bit Linux OS?  There was a problem with old "387" 80bit floating point math that was fixed just a couple of days ago - issue #3313 .  I recommend you upgrade to the very latest code (see the FAQ for upgrade instructions).

-Kevin

mistrjirka, 2020-09-25:
I did. Thank you. I upgraded OS and also downloaded new klipper. If anything comes up. I will reopen the issue.

closed 2020-09-25 by mistrjirka