Klipper3d/klipper #3288: MKS Gen l V2.1 pinout error bltouch

makuproyect, 2020-09-06:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5179278/klippy.log)
[printer.log](https://github.com/KevinOConnor/klipper/files/5179279/printer.log)

Dear your support to be able to configure the printer.cfg file for a MKS GEN L V2.1 board
It does not recognize pin 11, 21, 39, 32 (where the BLtouch will go) I have guided the number of pins from https://github.com/makerbase-mks/MKS-GEN_L/blob/master/hardware/MKS%20Gen_L % 20V2.1_001 / MKS% 20GEN_L% 20V2.1_001% 20PIN.pdf

I appreciate your help.

Estimados su apoyo para poder configurar el archivo printer.cfg para una placa MKS GEN L V2.1
no reconoce el pin 11, 21, 39, 32 (donde ira el BLtouch) me he guiado el numero de pines de https://github.com/makerbase-mks/MKS-GEN_L/blob/master/hardware/MKS%20Gen_L%20V2.1_001/MKS%20GEN_L%20V2.1_001%20PIN.pdf

Agradesco su ayuda.

makuproyect, 2020-09-06:
Se actualizo klipper a su ultima versión el problema persiste puede ser alguna incompatibilidad con la motherboard?
se valida que en marlin los pines de bltouch son: 
sensor_pin: ar18
control_pin: ar11
Agradezco su pronta ayuda.

KevinOConnor, 2020-09-11:
If you wish to use the arduino aliases, you must add `pin_map: arduino` to the `[mcu]` section of your config - see config/example.cfg for details.

-Kevin

makuproyect, 2020-09-16:
Thanks, problem solved.
Gracias, problema solucionado.

closed 2020-09-16 by makuproyect